S7NetPlus/s7netplus #20: Performance estimate?

ErikWahlstrom, 2015-01-21:
Sorry to post here but did not find anywhere to post general questions.

Anyway, do you have an estimate on how long a read cycle and write cycle of a byte in the PLC will require? Does it vary a lot and if so do you have any ideas as to why?

I am currently using a OEM library from Siemens to read and write to the PLC but the communication speed is not that good. We are using a Sinumerik system (I have pretty poor knowledge on the PLC/NC side of things but I can specify in detail if needed) which isn't as fast as we'd like it to be. We have used S7.net prior for test-purpose so I know it is possible to use it for our system. Just wanted to know what to expect before starting to investigate any further on my own.

killnine, 2015-01-21:
To be honest, I haven't done any real benchmarking with S7Net to see how it performs. For me, it's always been 'fast enough'. What performance metrics are you looking at where your current solution isn't sufficient?

ErikWahlstrom, 2015-01-22:
For us now a read cycle can vary from 10 milliseconds up to 100 millseconds or worse. With the current API we can read many variables during the same read but the amount of variables/data does not affect the cycletime that much. We have a vision system on the PC (PCU with windows) which needs to react as soon as possible. We are communicating via variables to start a vision command. The communication requires a couple of read and write cycles of bool and int variables. Sometimes the whole communication cycle from PLC to Vision requires more than one second which is too long to be acceptable.

mesta1, 2015-01-26:
To optimize the read-write time you need to know how the protocol works, the less requests you make, the faster the communication is.
Let's say that the fastest approach is to save all the HMI values in a single DB, then read the DB as bytes, then make all the conversions by hand.
Another fast approach is by using the ReadClass feature, but there is some overhead due to the reflection that is used to retrieve the variables.
It's much faster to read 220 bytes in a single DB than 20 bits in 20 different places.

killnine, 2015-01-26:
Exactly. We generally create 'readblocks' where we consolidate key data
into a single place to grab it all at once. We also version the read blocks
so, as the program matures, we can parse it correctly on the back end.

There's a fixed amount of overhead, as Mesta noted, so there may just be a
floor at which you really can't retrieve much faster. 10-100ms is probably
within the margin of networking latency, too, for that matter.

ErikWahlstrom, 2015-01-26:
Thanks for the replies!

The current library has the ability to read/write chunks of data in one operation which executes almost as fast as reading just single values. We use this whenever it is possible but for this particular problem it's not an option. We have to do some handshaking to make sure the operation (Vision evaluation) starts, executes and gets a result. So typically we listen for a variable  to change (polling all possible variables at once) and then we have to read some data which corresponds to which parameters should be used by the Vision system. And then when the evaluation is done we need to tell the PLC it's done. So these read/write operations has to be in a sequence. Sometimes we use up to one second just for the communication data exchange which causes stop in the overall automation.

I'll probably go ahead and see what can be achieved s7netPlus but I wanted to know if it was a waste of time or not :)

I'll let you know the result if/when I've had time to test this.

jogibear9988, 2015-02-25:
For Information:

Every Read Request to a CPU can be PDUSize - 18 Bytes wich is a header. It does not matter wich Library you use (this one, libnodave, mine,...). The perfomance bottleneck mostly will be the CPU wich need's time to answer the request. From more different areas you read, less bytes you can read at once. For each request to another area, you have to reduce the read size with 6 bytes, because this is the header for the request. That means, if you read in a DataBlock, but you have hole of 4 Bytes you woould not read, it would be better to read them, because this will result in less read requests!

Maybe also this helps someone: http://www.bj-ig.de/140.html

closed 2015-09-27 by mesta1