S7NetPlus/s7netplus #24: This library has no support for String data type

mergoktas, 2015-05-18:
This library has no support for String data type although  it is defined as VarType. It is not possible to read or write class or structure containing string data. Am I wrong?  There is no case in switch staments for "String" data type for converting from byte array to struct/class in source codes. Otherwise how?

Besides, comments documentation like this 
        /// <summary>
        /// Gets the size of the struct in bytes.
        /// </summary>
        /// <param name="structType">the type of the struct</param>
        /// <returns>the number of bytes</returns>
        public static int GetClassSize(Type classType)
{....

should be corrected, because it is a class type not a struct type for 
public static class Class

adream78, 2016-02-18:
Hi,
news about string support ?
thanks

pw-ffx, 2016-02-18:
I'm reading strings with it. I had to fix a small problem concerning string termination and string length.

movAX13h, 2016-02-18:
Strings should work. I had to fix some small issues concerning string termination and string length but that was more than 2 years ago.

adream78, 2016-02-18:
Hi,
in PLc.DB1.cs I have
        public bool Flag00 { get; set; }
        public bool Flag01 { get; set; }
        public bool Flag02 { get; set; }
        public bool Flag03 { get; set; }
        public bool Flag04 { get; set; }
        public bool Flag05 { get; set; }
        public bool Flag06 { get; set; }
        public bool Flag07 { get; set; }
        public string mystring { get; set; }

but 

Plc.Instance.Db1.mystring return null
other bool and byte variable are all tested.

In my Siemens S7-300 I have a DB1 with bool value and a STRING[50]

where can I define the lenght in s7netwrapper ?

In Visual studio 2013 I have download from nutget packages manager s7.netplus 0.1.1 version.

Please can you help me?
thanks

movAX13h, 2016-02-18:
I've never used the method to read complete structs since the variables I have to read are not relieable in a sequence (addresses). In PLC.cs you can see that strings are supported. Maybe it helps to know that strings are just byte arrays. I've found out that the first 2 bytes contain the string length (S7-300) and there is no termination byte. I've modified the lib for my purposes - custom syntax for reading all types (because the Siemens syntax is not very flexible, for example for string length without reading the 2 bytes first), automatic sequence-finding for all required variables to reduce the number of requests. What I wanted to say: Just look into the classes, it's not so difficult to understand what it is doing and extend it to your needs. 

I'm sorry, I dont have time to look into the current codebase at the moment and figure out if there is an immediate solution to your problem.

tanner-wood, 2016-02-19:
I'm reading strings fine on 300, 1200, and 1500 units. I've found that I
need to read twice as many bytes as  expected to get the full string.

FriedrichWedel, 2016-03-23:
There are STRING[x] and WSTRING[x] types.

At STRING type: First byte is max. length. Second byte is current length. Then the ASCII characters are following.

At WSTRING type: Exactly like STRING but you have two bytes for everything, First UINT is max. length,; sencond UINT is current length then the WCHAR characters are following.

I think at least the Write should update the current length BYTE/UINT if the string will be used within the S7 PLC.

GROMFU, 2018-07-12:
Can someone provide a valid code example (structure declaration in c# and how to call a method via s7netplus) in order do show how to handle strings **inside** a structure ? Please :)

mycroes, 2018-07-12:
@GROMFU this is currently not supported. If you want to discuss your problem, open a new issue (this issue is about string support in general).