S7NetPlus/s7netplus #33: Reading / Writing DB1 block

Fordirk, 2015-10-05:
Hi There !

I have done a few projects using this codebase and mostly it works pretty well. I have been able to read / write all tags with Markers So anything like MW20 or M5.3 is no problem. I have not been able to address Datablocks however. DB1.DBX.0.1 for example. I'm using a S71214. Is it an addressing issue on my part ? Could you possibly help me out ?

Kind Regards,
Fordirk

oiisamiio, 2015-10-05:
you mean DB1.DBX0.1 right?

Regards, Sam

killnine, 2015-10-05:
@Fordirk Can you provide some sample code so we can look at how you're making the call?

Fordirk, 2015-10-07:
Thank you for your reply. Here's how the project starts. The Line that writes MW20 works ok.
            DB1.DBX0.1 (thanks Sam) throws errors. in the PLC config i have Data_Block_1[DB1],    containing 4 Static Booleans. I'm no PLC specialist. Is this correct ?

           //At button connect inform the user that there is contact with the S7 PLC
            try
            {
                listBox1.Items.Add("Checking PLC");
                this.Update();

                Plc.Instance.Connect(Properties.Settings.Default.PLC_IP);  // connect to plc 
                listBox1.Items.Add("PLC OK");
                listBox1.BackColor = System.Drawing.Color.White;
                Plc.Instance.Write("MW20", int.Parse(110));  // default Scandelay  (works OK)
                Plc.Instance.Write("DB1.DBX0.1", true);  // Always throws WrongVarFormat
                this.Update();
                // use plc.instance.Disconnect to sever communications

            }
            catch (Exception exc)
            {
                listBox1.Items.Add(" !!  PLC not connected or address error !! ");
                listBox1.Items.Add(exc.Message);
                listBox1.BackColor = System.Drawing.Color.Pink;
                this.Update();
            }

oiisamiio, 2015-10-07:
value should be 1 (for true) or 0 (for false)

killnine, 2015-10-10:
There's a couple places in Write() where this might be able to be improved. 

I don't have the code running on this machine but I think one problem is the method @oiisamiio calls tries to cast his boolean to an integer before passing it to another Write() call. That could probably handle the boolean case more gracefully (since it does make sense that you may want to pass in true/false rather than 1/0).

Furthermore, the overload of the Write method doesn't have a case statement to support the boolean type.

ozcanay, 2016-12-08:
Hi,

I am having the same issue as well. I am also using S7-1214. I was able to read any values in DB1. I have tried reading double, bool, int value, but no success. Do anyone have  any idea about this issue? May it be a problem on 1214 model?

Thanks in advance.

oiisamiio, 2016-12-08:
Can you also provide some sample code?

ozcanay, 2016-12-08:
My code is actually short : 

`using System;
using System.Collections.Generic;
using System.Linq;
using System.Text;
using S7.Net;

namespace ConsoleApplication1
{
    class Program
    {
        static void Main(string[] args)
        {
  
            String MyCpuIp = "192.168.0.101";
            short MyCpuRack = 0;
            short MyCpuSlot = 1;

            using (var plc = new Plc(CpuType.S71200, MyCpuIp, MyCpuRack, MyCpuSlot))
            {
                    if (plc.IsAvailable)
                    {
                    plc.Open();
                        if(plc.IsConnected)
                        {
                            Console.WriteLine("PLC is Available and Connected");
                        
                        bool a = (bool) plc.Read("DB2.DBX0.0");
                        Console.WriteLine(a);
                        
                    } else
                        {
                            Console.WriteLine("PLC is Available but NOT Connected");
                        }
                    }
                    else
                    {
                        Console.WriteLine("PLC is NOT Available");
                    }
                plc.Close();
;               Console.ReadKey();
            }

        }
    }
}
          `

I have written to Siemens Forum as well : 

https://support.industry.siemens.com/tf/ww/en/posts/reading-datablock-with-s7-net/158606/?page=0&pageSize=10#post644539

oiisamiio, 2016-12-08:
Set an Breakpoint like this:
[image]

what exception will raised? Or Breakpoint not will be reached?

ozcanay, 2016-12-17:
The problem was actually CPU firmware not being up-to-date. It worked with PLCSIM.

ozcanay, 2016-12-17:
[image]

Hi,

I am trying to read DB1.DBW0.0 "continuosly". So, I created a timer for Windows Form application which reads the address every 100ms. At first reading the read value is correct. After couple(I guess) of readings reading process will be broken and textbox displays "Problem reading var" as it is stated in the code. There is a problem with the cast exception. I have no idea why it will work at first then.

Thanks in advance.

oiisamiio, 2016-12-20:
Hi, first excuse my english

setup breakpoints in plc class catch exception block to specify your problem.

ozcanay, 2016-12-20:
I figured the problem out. I didn't write plcWriter.Close() at the end of the code. Oops!

mesta1, 2016-12-21:
You can also use plc.Open() once at startup of application, and plc.Close() at the end of the application.