S7NetPlus/s7netplus #35: Read real value from PLC

ramanandmittal, 2016-01-18:
Dear team,

We are not able to read real value from this program. Please share help for this.

oiisamiio, 2016-01-19:
Hi there,

reading real works. Just precision looks dirty...

Whats the problem? You don't wrote additional Informations.

[image]
[image]
[image]

catenarysoftware, 2016-01-22:
I too have been trying to read real data types. I see in your screen shots you have a Bit.cs class, is this a new class that has not be committed yet to Github?

mesta1, 2016-01-22:
This is an example on how to write and read REAL values. You can check it on  S7.Net.UnitTest / S7NetTests.cs.
Basically you have to convert the number in dword to write it, then convert to real after read it. S7.Net provides all the extensions you need to do this conversions.

// Reading and writing a double is quite complicated, because it needs to be converted to DWord before the write, 
// then reconvert to double after the read. 
double val = 35.687; 
plc.Write("DB1.DBD40", val.ConvertToUInt()); 
double result = ((uint)plc.Read("DB1.DBD40")).ConvertToDouble(); 
}

ramanandmittal, 2016-01-28:
Thanks dear We are able to read real values

closed 2016-01-28 by ramanandmittal

CFish66, 2023-01-09:
Hi everybody,

I'm a new in VS and S7.Net, In my case, i got an error when reading Real value.
Below is my code, it cannot use ".ConvertToUInt()". C# display and error. I try to translate
from Chinese to English. It mean 'Double' not included definatiion of ".ConvertTUInt()'.

For write a real value to PLC. My case can write a float value directly to plc and plc value is correct.

private void button2_Click_1(object sender, EventArgs e)
        {
            double val = 35.687;
            uint val_1;
            float aa;

            //val_1 = (double)val.ConverToShort();
            Globals.PLC_Struct[5].plc.Write("DB1.DBD0", val.ConvertToUInt());

            //var converted = Convert.ToUInt32(val);
            //Globals.PLC_Struct[5].plc.Write("DB1.DBD0", (uint)val);

            //float dword = Convert.ToSingle( Globals.PLC_Struct[5].plc.Read("DB1.DBD0"));
            //double result = Convert.ToDouble(dword); ;
        }

CFish66, 2023-01-09:
Does someone can help in reading real value ?