S7NetPlus/s7netplus #36: Universal Windows Platform support (iot)

watashimeandeu, 2016-01-26:
Hi

I already test this lib on windows 10 x32, work very nice, but on windows 10 core (raspberry pi) don't allow me to install it by nuget.

It's possible to build an uwp version?

Thank you

watashimeandeu, 2016-01-29:
With few tests i thing already working.
Thanks to [this code](http://www.codeproject.com/Articles/22918/How-To-Use-the-SocketAsyncEventArgs-Class) i've changed on PLC.cs line 15

`private Socket _mSocket;`
to
`private SocketAsyncClient.SocketClient _mSocket;`
and use this code on new class file:

	using System;
	using System.Net;
	using System.Net.Sockets;
	using System.Threading;

	namespace SocketAsyncClient
	{
		/// <summary>
		/// Implements the connection logic for the socket client.
		/// </summary>
		internal sealed class SocketClient : IDisposable
		{
			/// <summary>
			/// The socket used to send/receive messages.
			/// </summary>
			private Socket clientSocket;

			/// <summary>
			/// Flag for connected socket.
			/// </summary>
			public Boolean connected { get; set; } = false;

			/// <summary>
			/// Listener endpoint.
			/// </summary>
			private IPEndPoint hostEndPoint;

			/// <summary>
			/// Signals a connection.
			/// </summary>
			private static AutoResetEvent autoConnectEvent = new AutoResetEvent(false);

			/// <summary>
			/// Signals the send/receive operation.
			/// </summary>
			private static AutoResetEvent sendEvent = new AutoResetEvent(false);
			private static AutoResetEvent receiveEvent = new AutoResetEvent(false);

			/// <summary>
			/// Create an uninitialized client instance.  
			/// To start the send/receive processing
			/// call the Connect method followed by SendReceive method.
			/// </summary>
			/// <param name="ip">IP of the host where the listener is running.</param>
			/// <param name="port">Number of the TCP port from the listener.</param>
			internal SocketClient(String ip, Int32 port)
			{
				// Addres of the host.
				IPAddress address = IPAddress.Parse(ip);

				// Instantiates the endpoint and socket.
				this.hostEndPoint = new IPEndPoint(address, port);
				this.clientSocket = new Socket(this.hostEndPoint.AddressFamily, SocketType.Stream, ProtocolType.Tcp);
			}

			/// <summary>
			/// Connect to the host.
			/// </summary>
			/// <returns>True if connection has succeded, else false.</returns>
			internal void Connect()
			{
				SocketAsyncEventArgs connectArgs = new SocketAsyncEventArgs();
[... 133 lines cut ...]

			/// <summary>
			/// Disposes the instance of SocketClient.
			/// </summary>
			public void Dispose()
			{
				autoConnectEvent.Reset();
				sendEvent.Reset();
				receiveEvent.Reset();

				autoConnectEvent.Dispose();
				sendEvent.Dispose();
				receiveEvent.Dispose();

				if (clientSocket.Connected)
					clientSocket.Shutdown(SocketShutdown.Both);

				clientSocket.Dispose();
			}

			#endregion
		}
	}

Hope this help.

mesta1, 2016-01-30:
Nice find, but I don't think we will support Universal Apps for now.

watashimeandeu, 2016-02-04:
No problem. 
I made some changes and added one pull request to contribute.

Thank you.

mesta1, 2016-02-04:
I will look into your pull request, but this will take time because you are replacing the whole .Net socket class with your own.

watashimeandeu, 2016-02-05:
I know, but after several tests with S7 1200 (in my case) it was the only way to get successful results.
Take as much time as you need.

mesta1, 2016-02-08:
To support this we can't just merge your changes, because they are breaking changes, basically:
1) solution is limited only to VS 2015
2) untested socket class
What we can do is to extract the common parts in a common project, let's say "S7Net.Common",
 then create 2 solutions:
1) a VS2010 compatible and uses the standard .Net 3.5 socket
2) a VS2015 .Net Core compatible solution with the custom socket
This is quite some work to do, but if you can do it, that would be awesome. If you can't, keep the pull request and i'll try to do it when I'll have some spare time.

watashimeandeu, 2016-02-08:
You're right.
I can do it. 
I will remove the current pull request and add new one as soon as i complete the task.

watashimeandeu, 2016-02-08:
Ups. It's not a solution, it's a project, but i think will work if not included all at one.

mesta1, 2016-02-08:
I will fix it.

mesta1, 2016-02-08:
I created a separate solution for Universal Windows Applications.

closed 2016-02-08 by mesta1