S7NetPlus/s7netplus #37: Connect S7 PLC through MPI Adaptor

ramanandmittal, 2016-01-27:
Hi team,

Can you please let me know how we can make connection string fro MPI adaptor.

killnine, 2016-01-27:
That's a good question but unfortunately I haven't got the hardware to test communication with an MPI adapter. Is there a way you might be able to make a 'staging area' on your CPU that has a copy of the remote data coming over from the MPI adapter?

oiisamiio, 2016-01-27:
S7NetPlus is based on Socket Connection via Tcp. Its not possible to connect via MPI. Try libnodave or https://github.com/dotnetprojects/DotNetSiemensPLCToolBoxLibrary based on libnodave...

closed 2016-02-08 by killnine

ramanandmittal, 2016-02-09:
Can you provide MPI adaptor connection string So i can check with MPI through this code