S7NetPlus/s7netplus #38: Problem with DB size bigger of 8200 byte

unocelli, 2016-01-30:
Hi team,
can you please let me know how we can read one DB with a size bigger of 8200 byte?

I read DB with ReadBytes(DataType.DataBlock, 1, 8200, 200);
I saw that the problem is by "package.Add(Types.Word.ToByteArray((ushort) ((startByteAdr)*8)));"

Best Regards

oiisamiio, 2016-01-30:
https://github.com/oiisamiio/s7netplus/pull/1

https://github.com/jobmeij/s7netplus/commit/f6246382a862040341d1f8626a2a6556fe28609d

unocelli, 2016-02-01:
Hi and tanks...maybe is better?
package.Add((byte)(startByteAdr / 8191));

Best Regards

oiisamiio, 2016-02-02:
Sorry your code looks dirty and will result in trouble.

an example:
If you start read at address 8100 and set read 200 bytes like
ReadBytes(DataType.DataBlock, 1, 8100, 200);
data will be corrupt because we cant split actually streamaddress.

so we should check startaddress lower than 8191 and count + startaddress greater than 8191

Excuse my English...

Regards Sam

unocelli, 2016-02-02:
Hi Sam,
What do you do if you have an array of 40000 bytes? after  ~16200  you have again a overflow? (~24000 again)...?

Excuse my English...
Regards Umberto

oiisamiio, 2016-02-03:
Think so. How many bytes you want to read?

unocelli, 2016-02-03:
Hi Sam,
a DB in a S7-300, S7-400, S7-1200 PLC can have 64K size. I think is a
quality question...and i come up with the problem because i'm compared with
this. i don't know the correct solution but your bug fixing is more wrong
as my...i tested.

Best Regards
umberto

closed 2016-02-13 by mesta1