S7NetPlus/s7netplus #44: Missing method ClearLastErrorCode

FriedrichWedel, 2016-03-02:
Currently once the LastErrorCode is set e. g. because of a read error you have to disconnect and connect again to continue.

I would like to poll different variables from the S7 plc periodically and avoid to reconnect after a bad ReadBytes call. Is it possible to add a ClearLastErrorCode method for that? Thank you!

killnine, 2016-05-02:
I'm interested to know if this logic was put in by juergen1969's original implementation because the connection was somehow unrecoverable after a read error. I don't have an S7-1200 to test on anymore, so it's tough for me to continue development.

However, implementing a more robust retry/reconnect logic seems to be a more permanent solution.

I wonder what the community thinks...

mesta1, 2016-05-02:
Why do you have to disconnect and reconnect to continue? It just catch the exception and handle it, so to me it seems that you can continue the communication with the plc.
Of course if the ReadBytes is wrong due to connection problems, of course you have to connect and disconnect, but this is not dependant on how we handle the errors.
Error handling is not the best one, but it isn't the biggest problem at the moment.

FriedrichWedel, 2016-05-09:
@mesta1: I am talking about errors which may occur if you are reading e. g. a datablock register which is not defined (or out of scope: if you read register 2000 but the last byte in the datablock is on register 1000). I get this error on a S7-1200, but I am sure the S7 firmware is doing similar checks on other Plcs as well.

The exception is catched internally within ReadBytes and the LastErrorCode is set. The public property LastErrorCode currently tells me that an error occurred during the last read operation. Now it would be great if I can only clear this "LastErrorCode" flag and continue with e. g. reading/writing some other operation to get notified about the success/failure of the next read/write operation.

@killnine: 
The only code changes I wish would be to add a new method (no side effects and downward compatible):
File PLC.cs

public void ClearLastError()
{
     LastErrorCode =ErrorCode.NoError;
     LastERrorString = null;
}

mesta1, 2016-05-09:
Ok I will add it later.

closed 2016-05-11 by mesta1