S7NetPlus/s7netplus #47: Universal Windows Platform - How to?

Tycos, 2016-04-10:
Hi,

I tried S7.Net example and it just worked great, but i saw it is possible to run it also on a raspberry pi using UWP.
I found the S7.Net.Core project but i am not sure how to apply this.

I created a new project, copied the S7.Net.Core folder into the project folder and added it to the project. But i dont know how to continue :/

watashimeandeu, 2016-04-10:
I'm not sure if i really understand the difficult.
It's exactly the same.
You can compile the project and use the S7.Net.Core.dll as reference on your uwp project.
To raspberry pi compile as ARM, to local machine, x86 it's ok.

Tycos, 2016-04-10:
I am pretty new to this hole thing, so it was easy to me to understand a existing project, but creating a own one and set the right references is way harder.

watashimeandeu, 2016-04-10:
When you click "debug" dll files are created on "bin" folder by default. In your project you can right click and "Add References" and chose the dlls you need.
I think this may help you: http://ms-iot.github.io/content/en-US/GetStarted.htm.

Tycos, 2016-04-10:
Currently i try to get a connection like the README.md says. My problem is i am not sure how to define the PLC right. ---> [http://snag.gy/5Mk87.jpg](url)

[https://www.dropbox.com/s/awn41p9dvz917mt/RaspberryHMI.rar?dl=0](url)

watashimeandeu, 2016-04-12:
C# is case sensitive.
Try this changes:

	//Basic connection properties
	string deviceIpAddress = "192.168.0.178";
	short rackNumber = 0;
	short slotNumber = 0;

	//Connection to device
	using (var plc = new Plc(CpuType.S71200, deviceIpAddress, rackNumber, slotNumber))

Tycos, 2016-04-12:
Thanks, you are right.
i should have been able to solve this by myself ;p

Now its time to learn much more about C#

closed 2016-04-12 by Tycos