S7NetPlus/s7netplus #49: ReadMultiVars

DJWassink, 2016-04-16:
I was wondering if there is a way to read data from different location at once. Of course it is faster to prepare a DB or a UDT and read it in one go, but since I can't modify the current program of the PLC to achieve this I am looking for a quick way to read different locations.

The only options right now seems to use the Read methode repeatability but sadly enough that is too slow for my usecase, is there already something in place to do this?

Good job on the library it's very nice!

oiisamiio, 2016-04-16:
use async multiple connections to speed up your reads.

An example CPU 317 Supports up to 16 connections...

mesta1, 2016-04-16:
If you can implement the function by yourself, we accept pull request.
Here the documentation : http://www.bj-ig.de/service/verfuegbare-dokumentationen/s7-kommunikation/aufbau-eines-data-tpdus/parameter-part/index.html
If you check the "Read" paragraph, you see that you can ask for multiple area in the same request.

DJWassink, 2016-04-17:
Thanks both of you for you reply, reading from multiple connections is fast enough for now so thans for that answer.

However in the near future I am going to take a look at implementing the function. Thanks a ton mesta1 for the documentation really useful!

DJWassink, 2016-04-20:
I am currently trying to implement this feature, however I am running against a issue which I just can't get my head around. The current result returned has the different variables intertwined.

The current result of a read of 2 different variables ( and ) results in the following byte buffer returned from the s7server

```
[0]	3	byte
[1]	0	byte
[2]	0	byte
[3]	32	byte
[4]	2	byte
[5]	240	byte
[6]	128	byte
[7]	50	byte
[8]	3	byte
[9]	0	byte
[10]	0	byte
[11]	0	byte
[12]	0	byte
[13]	0	byte
[14]	2	byte
[15]	0	byte
[16]	11	byte
[17]	0	byte
[18]	0	byte
[19]	4	byte
[20]	2	byte
[21]	255	byte
[22]	4	byte
[23]	0	byte
[24]	8	byte
[25]	64	byte
[26]	129	byte // expected to be 0
[27]	255	byte
[28]	4	byte
[29]	0	byte
[30]	8	byte
[31]	0	byte
[32]	0	byte // expected this one to be 129
```

So basically all the values I get with the read method get set in the first part of the returned data area. Someone has any idea why this is happening and how to solve it?

To see the code I am currently working on the following branch; https://github.com/DJWassink/s7netplus/tree/feature/read-multiple-vars

Sadly I can't test it with a real PLC so I am not completely sure if it is s7server which is returning a wrong result, but I doubt its snap7 it's fault.

watashimeandeu, 2016-04-22:
On PLC.ReadMultipleVars(List) try:

	int offset = 25;
	foreach (var dataItem in dataItems)
	{
		int byteCnt = VarTypeToByteLength(dataItem.VarType, dataItem.Count);
		byte[] bytes = new byte[byteCnt];

		for (int i = 0; i < byteCnt; i++)
		{
			bytes[i] = bReceive[i + offset];
		}

		offset += byteCnt + 4;

		dataItem.Value = ParseBytes(dataItem.VarType, bytes, dataItem.Count);
	}

DJWassink, 2016-04-22:
Aah thanks I remember I still had to add support for reading variables with a bigger byte size then 2, thanks a ton for remembering me + the solution :+1:

mesta1, 2016-05-01:
Do you have any news on this ? Is it working ?

DJWassink, 2016-05-01:
Sorry I haven't been active with it lately, mainly because I can't test it against a real PLC right now. But it does function yes, I managed to read a few different tags in one go perfectly.

I am still planning on splitting a single request with too many tags into multiple request. But sadly we don't have a way to determine the PDU size, so I got stranded there and haven't touched it since.

mesta1, 2016-05-01:
If you can create a pull request, maybe we can test and integrate it from here.

PR #53 (2016-05-01): Feature/read multiple vars
Reading of multiple variables in a single call. Re-factored a few methods (basically the Read method) to reuse a ton of code.

Seems like the PLC.cs in the core package still needs a update.

Will fix #49
closed 2016-05-01 by mesta1