S7NetPlus/s7netplus #50: Reading from address higher than 8096

Skokan44, 2016-04-26:
Hi,

I am trying to read a data from a DB on S7-1200 station. I ran into problem and so far wasn't able to solve it. Apparently there is some problem, when I am reading data from position higher then 8096. Is it possible or am I doing something wrong?

` ushort address = (ushort) (StackAdress + 2*RecPtr);
var o = _plc.Read(DataType.DataBlock, DataBlockIndex,address, VarType.Int, RecSize);`¨

Thanks,
Martin.

mesta1, 2016-04-26:
Hi, are you using the NuGet version ? 
We updated the sources and solved the problem, but we still have to update NuGet.
Try to download the latest sources and you still have the problem, let me know.

Skokan44, 2016-04-26:
Yes I was using the NuGet version, now it is working just fine. Thank you for your work!

oiisamiio, 2016-05-03:
var overflow = (int)(startByteAdr * 8 / 0xffffU); // handles words with address bigger than 8191

still results in invalid data if startByteAdr < 8191 + Count > 8191 and data overlap.
So better let requests result in an error if startByteAdr < 8191 but + Count > 8191?

mesta1, 2016-05-03:
Please can you post the code to reproduce the issue?

oiisamiio, 2016-05-03:
public ErrorCode Write(DataType dataType, int db, int startByteAdr, object value)

an example:
if startByteAdr would be 8190
and value length would be 4 bytes
var overflow = (int)(startByteAdr * 8 / 0xffffU); // handles words with address bigger than 8191
var overflow would result in 0
address would overlap, request would overlap 0/1. Right?

Excuse my english

mesta1, 2016-05-03:
I'll test it asap.

oiisamiio, 2016-05-04:
think we need something like this:

`` 
   if ((startByteAdr*8/0xffffU) == 0)
                {
                    if ((startByteAdr*8/0xffffU) + varCount > 0)
                    {
                        throw new Exception(ErrorCode.WrongVarFormat.ToString());
                    }
                }
``

mesta1, 2016-05-04:
I don't think that throwing an error is a solution, but I still had no time to check this.

mesta1, 2016-05-05:
I did some tests today, but I couldn't reproduce the problem.
This is the code that I tested;

` [TestMethod]
        public void T11_WriteDbd8190()
        {
            ushort val = 16384;
            plc.Write("DB2.DBW8190", val);
            var result = plc.Read("DB2.DBW8190");
            Assert.AreEqual((ushort)16384, result);
        }

        [TestMethod]
        public void T12_WriteDbd8190()
        {
            int val2 = 32000;
            plc.Write("DB2.DBD8190", val2);
            var result2 =  plc.Read("DB2.DBD8190");
            Assert.AreEqual((uint)32000, result2);
        }`

oiisamiio, 2016-05-09:
Sorry, did also some tests. Looks fine, no issue

Please Close Issue, Thanks

mesta1, 2016-05-09:
No problem, thanks for your time.

closed 2016-05-09 by mesta1