S7NetPlus/s7netplus #52: Missing Strong Name

SenenFernandez, 2016-04-28:
Hello =)

Please "**Sign the assembly**":

[image]

Cheers,
Senen =)

mesta1, 2016-05-01:
Is there any particular reason you need this ?

SenenFernandez, 2016-05-01:
https://msdn.microsoft.com/en-us/library/dkkx7f79.aspx

https://msdn.microsoft.com/en-us/library/wd40t7ad%28v=vs.71%29.aspx

https://bitbucket.org/scons/scons/wiki/CsharpBuilder

mesta1, 2016-05-01:
I tried to sign the assembly, then accepted a pull request, then it wasn't compiling anymore.
Is there something wrong ?
Actually I had to remove the snk.

killnine, 2016-05-02:
Does this have consequences in the future, though? Does this mean that @mesta1 has to compile every build from this point forward?

mesta1, 2016-05-02:
I think I did something wrong, because I did right-click on the project, signed the assembly and created a new snk file.
I believe I need to add a snk file at solution level, I will look into it on the next days.
However all opensource libraries in .Net out there are signed, so we should do this too, just in the right way.

FriedrichWedel, 2016-05-09:
For signing assemblies you need a code signing certificate which has to be ordered from somewhere an is valid only for a specific period (e. g. 2 years). If you do code signing with a auto-generated certificate (like within VS) this does only make sense for debugging  but not for production. 

@mesta1: Do you know an example open source project with an official code signing certificate?

mesta1, 2016-05-09:
Helix Toolkit (https://github.com/helix-toolkit/helix-toolkit) is signed, but I don't know if it's autogenerated or what.

FriedrichWedel, 2016-05-09:
Its an auto-generated snk file, maybe a auto-generated snk file does make sense in some use cases (I heard for COM interop or placing the DLL into the GAC it might be required)

mesta1, 2016-05-09:
I think that the main problem of not having a snk-file is that if you have an assembly that is signed and this assembly reference a non-signed assembly, the project will not compile.
I tried to sign the assembly with an autogenerated snk, the problem is that the first pull request that I accepted didn't compiled anymore.
The problem that I'm facing on adding a snk  is that it should compile after some pull requests, and it should be possible to be compiled by everyone.

SenenFernandez, 2016-06-10:
Fixed with commit f86281a =)

closed 2016-06-10 by SenenFernandez