S7NetPlus/s7netplus #56: Duplicated code and S7.Net.Core

mesta1, 2016-05-11:
I would like to unify the source files and use `#ifdef NETFX_CORE` to check if use the .Net socket or the custom one.
Also we can use #ifdef to define the correct behaviour for the properties IsAvailable and IsConnected.
Ideas ?

killnine, 2016-05-11:
Is this related to issue https://github.com/killnine/s7netplus/issues/29?

I'm not entirely sure why we're forced to use a custom implementation in the first place. UWP seems to support TCP sockets, just not between different apps.

mesta1, 2016-05-11:
Issue #36

mesta1, 2016-05-12:
@watashimeandeu What problems did you have on using the .Net sockets in .Net Core applications ?

watashimeandeu, 2016-05-12:
No problem, i think.
SocketClient is using .Net Socket but the idea was change the less code as possible on PLC class by creating the methods send and receive synchronous and returning the bytes transferred.
I use [this](https://msdn.microsoft.com/en-us/library/windows/apps/hh202858(v=vs.105).aspx) as example.

mesta1, 2016-05-12:
Ok I'll see what I can do with #ifdef.

mesta1, 2016-07-08:
@watashimeandeu please test .Net Core version.

watashimeandeu, 2016-07-11:
So far, so good.

closed 2016-07-13 by mesta1