S7NetPlus/s7netplus #58: ReadMultipleVars issue

watashimeandeu, 2016-06-08:
Reading multiple USInt vars only the first one read ok.
Adding the code bellow fix my problem, but not sure if it is the right solution.

PLC.ReadMultipleVars(List<DataItem>):

	int offset = 25;
	foreach (var dataItem in dataItems)
	{
		int byteCnt = VarTypeToByteLength(dataItem.VarType, dataItem.Count);
		byte[] bytes = new byte[byteCnt];

		for (int i = 0; i < byteCnt; i++)
		{
			bytes[i] = bReceive[i + offset];
		}

		offset += byteCnt + 4;

        //this code
		if (byteCnt == 1)
			offset++;

		dataItem.Value = ParseBytes(dataItem.VarType, bytes, dataItem.Count);
	}

mesta1, 2016-06-09:
Can you post the code to reproduce the problem also ?

watashimeandeu, 2016-06-09:
On Plc the vars are USInt type.

    public Dictionary<uint, byte> GetOptions(ushort db, uint[] offsets)
	{
		var result = new Dictionary<uint, byte>();
		var read = new List<S7.Net.Types.DataItem>();

		foreach (var offset in offsets)
		{
			read.Add(new S7.Net.Types.DataItem()
			{
				Count = 1,
				DataType = S7.Net.DataType.DataBlock,
				DB = db,
				StartByteAdr = (int)offset,
				VarType = S7.Net.VarType.Byte
			});
		}

		_cpu.Read(read);

		foreach (var item in read)
			result[(uint)item.StartByteAdr] = item.Value.ToByte();

		return result;
	}

mesta1, 2016-06-12:
I can't find USInt type on Step7 or on TIA Portal. Can you upload the plc program ?

watashimeandeu, 2016-07-07:
[image]

This should return: 1, 2, 3, 4 but return: 1, 8, 0, 4.

GrayDelacluyse, 2018-04-30:
I know this issue is very old, but for others searching, USint is 1 byte with a max value of 255.  Therefore, your c# class/struct should actually be Byte

mycroes, 2018-05-20:
This is probably fixed by a recently merged pull request. At least reading single bytes works now.

watashimeandeu, 2018-05-21:
Someone who tests this scenario just to be sure, and for me this issue can be closed.

mycroes, 2018-05-21:
I tested reading single bit / byte using ReadMultipleVars as mentioned in PR #117. I'd say close this, as the datatype in PLC doesn't seem to matter anyway.

closed 2018-05-21 by watashimeandeu