S7NetPlus/s7netplus #60: Stability issue - wrong bytes WrongNumberReceivedBytes

Skokan44, 2016-07-02:
I am using the library for continuous reading of data from 1200 PLC. Often I run into state, where I can not successfully open connection to the PLC because I keep getting wrong number of bytes in the first test.
`_mSocket.Send(bSend1, 22, SocketFlags.None);
 if (_mSocket.Receive(bReceive, 22, SocketFlags.None) != 22)
			    {
			        throw new Exception(ErrorCode.WrongNumberReceivedBytes.ToString());
			    } ` 

The problem is usually solved after some time and without my intervention. Any suggestion, what might be the issue?

mesta1, 2016-07-03:
Are you opening and closing the connection often ? Is the same issue than this : https://github.com/killnine/s7netplus/issues/29

I would start by shutdown the two sockets, as suggested by oiisamio.
`public bool Close()
{
    if (this._mSocket != null && _mSocket.Connected)
    {
        this._mSocket.Shutdown(SocketShutdown.Both);
        this._mSocket.Disconnect(false);
        this._mSocket.Close(100);
    }
    return this.IsConnected;
}`

Tell us if it works for you.

mesta1, 2016-08-28:
This has been fixed in version 0.1.4 on NuGet.

closed 2016-08-28 by mesta1