S7NetPlus/s7netplus #61: GetClassSize return zero

watashimeandeu, 2016-07-05:
Read class with four bool properties, for example, result on index out of range exception.
Adding, on Types.Class.GetClassSize before return, the code below works.
Just a workaround until bit types are pendent.

	if (numBytes < 2 && properties.Any())
		numBytes = 2;

mesta1, 2016-07-06:
This happens when there are less than 8 bools.
We can fix this, but for your convenience it's better that you declare 8 consecutive bools, even if you don't use four of them. This will match the DB memory structure and the size will be calculated correctly.

watashimeandeu, 2016-07-06:
Declare 8 bools, numBytes is 1. In ReadMultipleBytes result on invalid cast exception because is trying to convert byte to byte[].
It's correct to try cast to work with two types?

mesta1, 2016-07-07:
ReadMultipleBytes has a bug because the wrong function is used. This should work.
```

private List<byte> ReadMultipleBytes(int numBytes, int db, int startByteAdr = 0)
{
    List<byte> resultBytes = new List<byte>();
    int index = startByteAdr;
    while (numBytes > 0)
    {
        var maxToRead = (int)Math.Min(numBytes, 200);
        byte[] bytes = ReadBytes(DataType.DataBlock, db, index, maxToRead);
        if (bytes == null)
            return new List<byte>();
        resultBytes.AddRange(bytes);
        numBytes -= maxToRead;
        index += maxToRead;
    }
    return resultBytes;
}

```

watashimeandeu, 2016-07-07:
Yes, work.
Thank you.

closed 2016-07-07 by watashimeandeu

mesta1, 2016-07-07:
Please can you prepare a pull request ?