S7NetPlus/s7netplus #64: Read Class Issue

jonweaver, 2016-08-08:
While playing around I tried reading a DB into a class.  The DB only had 2 boolean values.  This caused the size measurement of the class to return 0 and it threw an exception for index being out of range.  

I added this before numBytes is converted to int:

`if (numBytes - Math.Round(numBytes) != 0)
              numBytes = Math.Ceiling(numBytes);`

Is that an appropriate fix for a DB containing only boolean that is less or more than a byte? 

`/// <summary>
        /// Gets the size of the struct in bytes.
        /// </summary>
        /// <param name="classType">the type of the class</param>
        /// <returns>the number of bytes</returns>
        public static int GetClassSize(Type classType)
        {
            double numBytes = 0.0;       
                            
            var properties = classType.GetProperties();
            foreach (var property in properties)
            {
                switch (property.PropertyType.Name)
                {
                    case "Boolean":
                        numBytes += 0.125;
                        break;
                    case "Byte":
                        numBytes = Math.Ceiling(numBytes);
                        numBytes++;
                        break;
                    case "Int16":
                    case "UInt16":
                        numBytes = Math.Ceiling(numBytes);
                        if ((numBytes / 2 - Math.Floor(numBytes / 2.0)) > 0)
                            numBytes++;
                        numBytes += 2;
                        break;
                    case "Int32":
                    case "UInt32":
                        numBytes = Math.Ceiling(numBytes);
                        if ((numBytes / 2 - Math.Floor(numBytes / 2.0)) > 0)
                            numBytes++;
                        numBytes += 4;
                        break;
                    case "Float":
                    case "Double":
                        numBytes = Math.Ceiling(numBytes);
                        if ((numBytes / 2 - Math.Floor(numBytes / 2.0)) > 0)
                            numBytes++;
                        numBytes += 4;
                        break;
                    default:
                        numBytes += GetClassSize(property.PropertyType);
                        break;
                }
            }
*****
            if (numBytes - Math.Round(numBytes) != 0)
              numBytes = Math.Ceiling(numBytes);
*****
            return (int)numBytes;
        }`

mesta1, 2016-08-09:
This isn't a bug, it is intended to let it crash in these cases. We need to improve the documentation on this case.

Siemens DB don't visualize the full byte when you use a boolean, so you better declare 8 bools in your class, rather than modify the sources.

While your code works on your case, it will not work with a DB like this:
DB0.DBX0.0
DB0.DBX0.1
DB1.DBX1.0
DB1.DBX1.1

Instead, declaring a class with 16 bools will work.

mesta1, 2018-05-13:
fixed https://github.com/killnine/s7netplus/pull/117

closed 2018-05-13 by mesta1