S7NetPlus/s7netplus #70: Interface for PLC

sohrt, 2016-11-15:
Hi,
I can see that there is a branch where a interface for the PLC have been added.
Is it by any chance possible to get that interface on the master branch, as it will be much easier to mock the PLC for unit testing?

killnine, 2016-11-15:
Ah, yes, that's a VERY old branch but would be a good idea for systems that depend upon this package.

I'll see what I can do.

mesta1, 2016-11-15:
The interface was removed to prevent this kind of situations. You should declare that interface.
To abstract the driver you should use the proxy pattern: http://www.dofactory.com/net/proxy-design-pattern
You should create your own class that implements your interface, and then use the driver in a private field and expose the methods that you need.
This is cleaner, because the unit tests will not need a reference to the S7.net driver and you have the maximum flexibility on extending your class and interface as much as you need.