S7NetPlus/s7netplus #73: Error while read Class/struct from s7-1200

meron1122, 2017-04-21:
Code
Issue -  index out of range exception  when read class.
````C#
namespace DemoS7NetPlus
{

    /// <summary>
    /// Interaction logic for MainWindow.xaml
    /// </summary>
    public partial class MainWindow : Window
    {

        public class TestDB
        {
            public bool TestowyBool { get; set; }
            public bool DrugiBool { get; set; }
        }

        public MainWindow()
        {
            InitializeComponent();

            Plc plc = new Plc(CpuType.S71200, "192.168.8.1", 0, 1);

            plc.Open();

            if (!plc.IsConnected)
                return;

            Task.Run(() =>
            {

        TestDB demo = new TestDB();

                while (true)
                {
                    plc.ReadClass(demo, 1); //here index out of range exception 
                    //"   w S7.Net.Types.Class.FromBytes(Object sourceClass, Type classType, Byte[] bytes)\r\n   w S7.Net.Plc.ReadClass(Object sourceClass, Int32 db, Int32 startByteAdr)\r\n   w DemoS7NetPlus.MainWindow.<>c__DisplayClass1_0.<.ctor>b__0() w C:\\Users\\Admin\\Documents\\Visual Studio 2017\\Projects\\DemoS7NetPlus\\DemoS7NetPlus\\MainWindow.xaml.cs:wiersz 55\r\n   w System.Threading.Tasks.Task`1.InnerInvoke()\r\n   w System.Threading.Tasks.Task.Execute()"

                    Thread.Sleep(TimeSpan.FromSeconds(0.25));
                }
            });
        }
    }
}
```

PLC SIDE:

![virtualbox_win7_32_tia 13_21_04_2017_11_35_28](https://cloud.githubusercontent.com/assets/2895090/25272060/af4d86b4-2686-11e7-8e09-663afc5f2d00.png)

mesta1, 2017-04-24:
You have to read at least 1 byte. 

related: https://github.com/killnine/s7netplus/issues/61

meron1122, 2017-04-25:
@mesta1 , Thank you, this make sense :)

mesta1, 2018-05-13:
Fixed https://github.com/killnine/s7netplus/pull/117

closed 2018-05-13 by mesta1