S7NetPlus/s7netplus #74: Error while reading string in class

meron1122, 2017-04-25:
I've got a class with String, String is declarated in struct on TIA too.
While ReadClass is executed, code throw Exception because String don't have empty Constructor.
So, how to read String property?

jkubisiowski, 2018-03-16:
Hey, try to read byte array and the convert it to string

mycroes, 2018-06-19:
Reading a `string` in a `class`/`struct` is only possible when the length is known upfront. I believe this is supported for strings in P/Invoke using attributes, so a solution would be possible. It's not really strange that it doesn't work out of the box though.

@meron1122: As you're mentioning the possiblity of string-in-struct in TIA, could you tell me if that string always includes the string header as well (1 byte size, 1 byte actual length)?