S7NetPlus/s7netplus #75: PLCSIM connectivity

wakemaster39, 2017-05-12:
Hello,

Sorry if this isn't on the s7netplus end, but I have been able to get the basics of this working with physical hardware, but when I attempt to connect to a PLCSIM instance of my s7-1200 I am unable to connect. 

I am trying a line similar to this `var plc = new Plc(CpuType.S71200, "127.0.0.1", 0, 1);` but when I run `plc.Open()` the last error is saying that I received back the wrong number of bytes. 

I am likely missing something in the TIA portal setup (using 14 SP1) but any help would be appreciated.

Thanks.

mesta1, 2017-05-12:
Here is the tutorial to configure NetToPlcSim: 
https://www.mesta-automation.com/nettoplcsim-how-to-connect-step-7-plc-sim-to-scadas/

wakemaster39, 2017-05-16:
@mesta1 Thanks for the response, I tried a number of things yesterday and searched the internet, the issue I am having is using TIA portal v14. Do you know a method of getting it working using that as the "PLCSIM s7-1200/S7-1500(TCP/IP)" configuration option in v13 to get it working does not exist.

mesta1, 2017-05-17:
You are probably requesting a wrong amount of bytes from the DB.
If the DB has 30 bytes, you can not ask for 60 bytes.

wakemaster39, 2017-05-18:
@mesta1 it was the put/get setting in the PLC configuration. Somehow it got reset and I haven't updated program on the PLC so it was still working. Thanks for the help.

closed 2017-05-18 by wakemaster39