S7NetPlus/s7netplus #80: How could I monitor one thousand or even more tags?

lvzhifeng2013, 2017-06-16:
Now,our solution is like this:Step 1,read the value of the tags;Step 2,compare the values with the last values that is read.Step 3,if the values equal the last values,it means that the values have not change;otherwise,it have changed.
       But  it takes about 5 seconds to read 1000 tags,if we need monior 10 thousand tags,it will be obout 1 minites.It is too long for us.Do you have a better way?

mesta1, 2017-06-17:
You should read block of bytes and extract single tags from them.
You can use ReadMultipleVars, but there are limits. I have to update the documentation on that regard. Here is sample code: 
https://github.com/killnine/s7netplus/blob/master/S7.Net.UnitTest/S7NetTests.cs#L385

lvzhifeng2013, 2017-06-19:
Maybe,your solution will word,but I don't thing it will solve the problem completely.I will try and test it this week,
      I am thinking that we define an event,when the value of a tag has changed,it will trigger the event.By doing this,we don't have to chech every tag each time.But I don't know how to write the trigger of the event.Do you think that my solution is possible?

allatrue, 2017-06-19:
The solution from @mesta1 will work.
If you wont to fire an event on value change, you should implement this trigger in PLC by yourself. So you implement some protocol on top of s7 standard protocol. For example you could monitor your 1000 values in PLC and set some small control area if some value changes. Your application polls just small control area and reads monitored values only if they change.

Note also, PLC needs time to process every read or write request. And this time depends on PLC type, communication processor or netlink adapter you use.

lvzhifeng2013, 2017-06-20:
I have just tested your solution. There are some problems:
A block may contains other DataType,such as string or bool, if I use ReadBytes,the result that is 
read will always be byte(it will change into the number on ASCII). Other DataTypes have the similar problems .
So, it turns out that using ReadBytes dosen't work.

Then, I try to use ReadMultipleVars,there are some other problems:
If the DataTypes of my 1000 tags contains several types, supose we don't known which DataType that each tag is and where the tag is. 
Step 1, I have to cherk all my tags and divide them into different types , in the meantime,because of the limit of ReadMultipleVars(less than 20 dataItems, and less that 200 Bytes ),I have to sum up the quantity of each DataType.
Step 2, divide them to some dataItems that each dataItems contains no more than 20 tags ,meanwhile, each dataItems can't reach to 200 bytes,so I check each dataItems.
Step 3, extract values from each dataItems and send values every tags.
This way seems so complicated and I have not tested how long it takes.

mesta1, 2017-06-20:
This is a low level library, it is not an opc server.

lvzhifeng2013, 2017-07-13:
I almost solved this problem,but when the DataType is bool or string, some problems come out.
How to use ReadMultipleVars to read several tags whose DataType is bool or string ?I know that I should use DataItem, but when it is bool,I found that you have not write this type in DataItem, when it is string it always  throw new Exception when the program goes to "if (bReceive[21] != 0xff) ".Can you teach me how to use ReadMultipleVars to read boll and string?

lvzhifeng2013, 2017-07-14:
I have just solved the bool problems. There is only one question: How to use  ReadMultipleVars to Read a couple of tags whose DataType are string?

lvzhifeng2013, 2017-07-19:
The limit of ReadMultipleVars(less than 20 dataItems, and less than 200 Bytes ),you said it is because of the limit of protocol.Is there some way to read more dataItems once?In the other word, could the package bigger and the package return is right?

mesta1, 2017-07-19:
No, you have to do multiple requests and to do them manually. Also the 200 bytes is an old limitation due to the https://github.com/killnine/s7netplus/issues/21
In the new plc you can get more than 200 bytes, but still you need to manually split bigger reads.

lvzhifeng2013, 2017-07-19:
Since I can not think out any other better solution,I will show my current solution to monitor 1000 or even more tags:
Step1,Add two member in dataItem "ID" , "Address" ,to return the value return belongs to whic tag.
 public class DataItem
    {

        /// <summary>
        /// Primary Key of the DataItem
        /// </summary>
        public string ID { get; set; }

        /// <summary>
        /// The address of the tag that you are about to read
        /// </summary>
        public string Address { get; set; }

        /// <summary>
        /// Memory area to read 
        /// </summary>
        public DataType DataType { get; set; }
        /// <summary>
        /// Type of data to be read (default is bytes)
        /// </summary>
        public VarType VarType { get; set; }

        /// <summary>
        /// Address of memory area to read (example: for DB1 this value is 1, for T45 this value is 45)
        /// </summary>
        public int DB { get; set; }

        /// <summary>
        /// Address of the first byte to read
        /// </summary>
        public int StartByteAdr { get; set; }

        /// <summary>
        /// Number of variables to read
        /// </summary>
        public int Count { get; set; }

        /// <summary>
        /// Contains the value of the memory area after the read has been executed
        /// </summary>
        public object Value { get; set; }

        /// <summary>
        /// Create an instance of DataItem
        /// </summary>
        //public DataItem()
        //{
        //    VarType = this.DataType;
        //    Count = 1;
        //}
        public DataItem(string id,DataType datatype,int db,string address,int startByteAdr,VarType vartype,int count)
        {
            this.ID = id;
            this.VarType =vartype;
            this.Count = count;
            this.DB = db;
            this.StartByteAdr = startByteAdr;
            this.DataType = datatype;
            this.Address = address;
           
        }
    }

Step 2: Let each tag to a dataItem and let every 19 dataItems to a list.

step 3:use ReadMultipleVars()
 for (int i = 0; i < list_count; i++)
                    {
                         plc.ReadMultipleVars(List[i]);    
                    }

step 4: using the value that is read by ReadMultipleVars() compare to the former value. If the value of one tag have changed, the program update the value of the tag,otherwise not.

If there are not string among 1000 tags,it takes about 300ms to read 1000 tag regardless of what varType each tag is.
If there is some string tags, I have to spit them first.For string tags ,I use Read(). For other varType, use ReadMultipleVars().How  long this case takes mainly depends on how many the string tags are.

mesta1, 2017-07-19:
Your solution is not optimized. If you have a lot of items in the same DB, you want to group them and make a single read in a single dataitem.

lvzhifeng2013, 2017-07-20:
@mesta1 Fist of all, I define an Array whose each element is a list. As you said, I  have to group some tags into a list, but I don't have to make a single read in a single datatem.For I have already grouped each tag  to  an exact list, so I have many list to read.Then I use ReadMultipleVars like this:
for (int i = 0; i < list_count; i++)
{
plc.ReadMultipleVars(List[i]);
}

so  I don't have to make a single read in a single datatem

mesta1, 2017-07-20:
If I remember correctly, the ReadMultipleVars can contain up to 20 dataitem. So at most if you are reading a single tag per data item, you read 20 bytes per request, which is not optimal.
DataItems are good in some cases, but the best is to read entire DBs and then extract the values that you need.

mycroes, 2018-06-19:
With S7-1500 the PDU size is 960 bytes. There's some bytes lost for the headers, but I believe you can read something like ~940 bytes in a single read that way (with proper PDU size negotation, which should work as of recently). When all the bytes have been received it's possible to extract data at the proper indices of the return array. I'm not sure about the timing of this, but I do know that reading 19 `Int16` variables (as separate `DataItem`s) takes 3.1ms with an S7-1500 (actually even less, because the benchmark used had to built the `DataItem`s in that time as well). @lvzhifeng2013: can you let us know if this solves your problem?