S7NetPlus/s7netplus #81: Read Array[0..5] of

SMOWiwam, 2017-06-21:
Hello,

I use the read Class and make my own array's in the constructor. The same size is inside the PLC
But I get an An unhandled exception of type 'System.MissingMethodException' occurred in mscorlib.dll
on Activator.CreateInstance(property.PropertyType);

SMOWiwam, 2017-06-21:
Visual Studio:
public class PC_Test
    {
        //public bool test { get; set; }
        public double[] testreal { get; set; }
        public PC_Test()
        {
            testreal = new double[6];
        }
    }
TIA Portal:
	testreal	Array[0..5] of Real	0.0			False	True	True	True	False

mesta1, 2017-06-21:
If it is a fixed array, you can probably declare them as 6 consecutive doubles.
ReadClass can't manage arrays at the moment.

mesta1, 2017-08-16:
This has been fixed in one of the last commits.

closed 2017-08-16 by mesta1