S7NetPlus/s7netplus #85: Read PLC Boolean Input Which is Not Connected

fkreuz, 2017-08-01:
Hallo,

I have a question regarding reading Boolean Input which is not connected from PLC, the result will be always TRUE. This is confusing because when the reading output is TRUE, we could not know for sure if it is really true or there is some connection problem. Below is my implementation:

```
using (var plc = new Plc(CpuType.S71500, ip_plc, 0, 1))
            {
                var result = plc.Open();

                if (result != ErrorCode.NoError)
                {
                    Console.WriteLine("Error PLC Connection: " + plc.LastErrorCode + " " + plc.LastErrorString);
                }

                while (true)
                {
                    bool db285TARGET_POS_REACHED = Convert.ToBoolean(plc.Read("DB285.DBX26.0"));
                    Console.WriteLine("DB285.DBX26.0: " + db285TARGET_POS_REACHED);
               }
}
```
When the PLC is connected, the above code works fine. But if the PLC is not connected or if the PLC connected but the "DB285.DBX26.0" is not available, the reading result will be always TRUE.
Are there any other ways for making sure that the reading process is successful? Thank you before.

Best, 
Finley

mesta1, 2017-08-01:
if you are offline, plc.Read("DB285.DBX26.0") will fail and raise an error. Check LastErrorCode  after every read.

fkreuz, 2017-08-02:
Hi @mesta1,

thank you for your answer. When my PLC is offline, the LastErrorCode after every read shows "Error PLC connection ...", but the "Read" command always produces TRUE value. 
Anyway I can still check the LastErrorCode first before reading. 

Thank you again.

closed 2017-08-02 by fkreuz