S7NetPlus/s7netplus #86: Can not read negative Values

Happe79, 2017-08-08:
Hello,

So far, thank you very much for that library. It helps me a lot.

But a have a problem reading negative values from DB.

E.g. I know that in the address area DB331.12.0 (DB331.Offset.Bit) is a values stored like -12 from Type "DINT", I get "-2147483648" using:

 object Result = spsPLC.Read(S7.Net.DataType.DataBlock, 331, iOffset, S7.Net.VarType.DInt, 1);
 long iNewValueInt = System.Convert.ToInt64(Result);
  sValue = "" + iNewValueInt;

return sValue;

If the values in the DB are positve it is working fine.

Do you may have an idea?

Thank you.

Stefan

mesta1, 2017-08-16:
System.Convert.ToInt64 is not correct.
You have to use the conversions provided by the library.

MoriLuca, 2018-02-27:
Sorry but i really cant get over this, i kindly ask for your help.

As far as i know, those are the Siemens Base data Types and the conversion that im tring to implement:

Siemens Int      ( 16bit )  -> DotNet Int16
Siemens UInt    ( 16bit )  -> DotNet UInt16
Siemens DInt    ( 32bit )  -> DotNet Int32
Siemens DUInt  ( 32bit )  -> DotNet UInt32
Siemens real     ( 32bit )  -> DotNet double     ( As i read from other post )
Siemens LReal   ( 64bit )  -> DotNet double    ( As i read from other post )

Im working on a proget based on Dot Net Core, building an asp net MVC Project.

While reading and casting a Siemens Int like this is working for positive and negative numbers:

```
object o = new object();
o = PLC.Read(S7.Net.DataType.DataBlock, 1, 0, S7.Net.VarType.Int, 1);
Int16 j = Convert.ToInt16(o);
```

For the double Int Negative Values this is not working :

```
object o;
o = .PLC.Read(S7.Net.DataType.DataBlock, 1, 4, S7.Net.VarType.DInt, 1);
Int32 j = Convert.ToInt32(o);
```

I read in some ohter post that i have to use the library methods for conversions, but i can not understand how to use them.
Sorry im a novice, but for the S7.Net.Conversion.ConvertToInt(), used for retrive negative numbers from DB, im expected to pass an uint value and i dont understand how to use it.

mesta1, 2018-02-27:
https://github.com/killnine/s7netplus/wiki/Value-conversion

MoriLuca, 2018-02-27:
Thanks for the answer, sorry im new to programming and to Git hub

mesta1, 2018-02-28:
Long real is not available. The code should be something like this:
```

  #region Get/Set 64 bit floating point number (S7 LReal) (Range of Double)
        public static Double GetLRealAt(byte[] Buffer, int Pos)
        {
            UInt64 Value = GetULIntAt(Buffer, Pos);
            byte[] bytes = BitConverter.GetBytes(Value);
            return BitConverter.ToDouble(bytes, 0);
        }
        public static void SetLRealAt(byte[] Buffer, int Pos, Double Value)
        {
            byte[] FloatArray = BitConverter.GetBytes(Value);
            Buffer[Pos] = FloatArray[7];
            Buffer[Pos + 1] = FloatArray[6];
            Buffer[Pos + 2] = FloatArray[5];
            Buffer[Pos + 3] = FloatArray[4];
            Buffer[Pos + 4] = FloatArray[3];
            Buffer[Pos + 5] = FloatArray[2];
            Buffer[Pos + 6] = FloatArray[1];
            Buffer[Pos + 7] = FloatArray[0];
        }
        #endregion

```

closed 2018-05-13 by mesta1