S7NetPlus/s7netplus #93: V0.1.7 sbyte write to DB don't work

SolarisD, 2017-09-15:
This code don't work with v0.1.7
sbyte sbMaxHW = 22; (signed 8-bit value - short int for PLC)
plc.Write(DataType.DataBlock, 1, iOffset, sbMaxHW);

but this code is working
byte bMaxHW = 22; (signed 8-bit value - short int for PLC)
plc.Write(DataType.DataBlock, 1, iOffset, bMaxHW);

mesta1, 2017-09-15:
Yes this is the 8-bit signed type. 
We still have to add support for that.

eltigre4jc, 2018-03-22:
Any news about this? I need also to write negative numbers to PLC..
Or is there any workaround possible?

SolarisD, 2018-03-22:
You shoud convert sByte to Byte and write Byte value to PLC

mycroes, 2018-06-19:
I guess that @SolarisD  means is cast the `sbyte` to `byte` and write the `byte` value. Would this be good enough for you @eltigre4jc?

mycroes, 2018-07-07:
Closing due to missing feedback. Feel free to reopen.

closed 2018-07-07 by mycroes