S7NetPlus/s7netplus #94: ReadClass() assigns all properties to 0 and false (SOMETIMES)

Libnerek, 2017-10-25:
I use this method in my program in a loop to scan all datablocks in my PLC and update mysql database. I created a control panel for my client this way. It works 99% of a time. From time to time it scans all the properties to "0" and "false". The next scan usually is successfull and stupid values are gone. But problem stays in charts. If chart update time matches with failed (0/false) scan it stays in a graphical representation and it looks bad. There is no anomaly for a certain datablock. All failed reads are totally random.
Do you have any clue what may cause it?

mesta1, 2017-10-28:
I have to investigate on this.

Libnerek, 2017-10-29:
Thank you for the answer! It made me feel more optimistic now... For the moment I will check a bit that supposed to be always 'true'. In case it is 'false' I will skip the connection.

mycroes, 2021-06-06:
This has been fixed for a while now, there's some checks in place to validate the read response.

closed 2021-06-06 by mycroes