S7NetPlus/s7netplus #105: Can't read structure. Help!!!

Mallowan, 2018-03-29:
Hello. I can't read structure. I've got an exception. 
I followed the instruction described in Documentation to th library. Please see screens 
[]( https://imgur.com/a/EHxha)

` public struct testStruct
        {
            public bool varBool0;
            public bool varBool1;
            public bool varBool2;
        }
        
        
            private void button1_Click(object sender, EventArgs e)
        {
            // reads a struct from DataBlock 1
            testStruct test = (testStruct)plc.ReadStruct(typeof(testStruct), 1);

            //textBox1.Text = new `testStruct().varBool0.ToString();``

Mallowan, 2018-04-11:
Hello there,
Of course I'll try do it this way. Thanks a lot.
But, as far as I understood "get" and "set" are used in declaration for a class reading. If we follow this documentation, for a structure reading isn't written that there should be used "get" and "set" in structure declaration. Isn't it? 
Either Documentation is poorly written or there's no difference between class reading and ctructure reading.

rapha-dev, 2018-04-11:
@Mallowan 
ReadStruct determines its size by field types (Boolean, Int32...) only.
ReadClass uses its properties (get, set) followed by field type to define final size.
In this case your testStruct should be fine without properties (at least contain 8 bools as @watashimeandeu recommends).

Mallowan, 2018-04-11:
Could someone share with working code of reading a structure to textBoxes and reading a structure to class with further output of values in textBoxes?

Mallowan

mesta1, 2018-04-11:
https://github.com/killnine/s7netplus/blob/9fd515280a4fe29ad41ce3d26eacdfd7866aeeb4/S7.Net.UnitTest/S7NetTests.cs#L211
https://github.com/killnine/s7netplus/blob/9fd515280a4fe29ad41ce3d26eacdfd7866aeeb4/S7.Net.UnitTest/Helpers/TestLongStruct.cs#L11

Mallowan, 2018-04-12:
Of course I saw these tests! If I got them I wouldn't write here my questions.
To me this is pretty complicate things to get such things on C# language.
I'm just young Step 7 programmer and in area of C# I have small experience. So I need examples.
I would be appreciate if someone would provide a code baced on real C# project.
**P.S.** I've found some sample projects in the Web.

rapha-dev, 2018-04-12:
Mesta has written about S7.Net in his [blog](https://www.mesta-automation.com/), an example is included, too. It's a good start to get familar with the library.

[Communication with Siemens S7 Plc with C# and S7.Net plc driver](https://www.mesta-automation.com/siemens-s7-plc-c-s7-net-plc-driver/)

Mallowan, 2018-04-12:
@rapha-dev,
First of all thank you for replying. I read this blog in the past and studied this sample. And as  I wrote above this example from Mesta is pretty poor. 
Context of events of provided sample  project:
-connection (start);
-disconnection (start);
-set Word variable;
-set Real variable;
-set DoubleInt variable;
-set DoubleWord variable;

There is no word about reading in the project code;
There's no word about a structure reading and a class reading in project code;
In the blog the class reading is positioned as a special feature of library, though.

If I'm not right, please correct me.

Mallowan

rapha-dev, 2018-04-19:
@Mallowan 

For my own i use classes to read from PLC and single statements to write.

A class read looks like this. `_plc` is a `S7.Net.Plc` instance.

```
        /// <summary>
        /// Read Class/Entry from PLC
        /// </summary>
        /// <param name="dsNumber">Entry no</param>
        /// <returns>False on read error</returns>
        public bool ReadFromDb(DB dbClass, int dsNumber)
        {
            if (_plc.IsConnected)
            {
                _plc.ClearLastError();
                _plc.ReadClass(dbClass, dbClass.DBNo, ((dsNumber - 1) * dbClass.StructLen) + dbClass.Offset);        // store result in dbClass-Instance
                if (_plc.LastErrorCode == ErrorCode.NoError)
                {
                    return true;
                }
            }
            return false;
        }
```

Each DB has its own class.

```
    /// <summary>
    /// Struct of DB800 entries
    /// PAR_LOG
    /// </summary>
    public class DB800 : DB
    {
        /***************************************************************************************
         * Must not contain other public properties than fillable by PLC maintaining order !!! *
         * aka "public Type propertyname { get; set; }"                                        *
         ***************************************************************************************/

        public DB800()
        {
            // set constraints
            Offset = 50;
            StructLen = 20;
            MaxLen = 2000;
            DBNo = 800;
        }

        #region Properties filled by PLC:ReadClass
        public Int16 Id { get; set; }
        public Int32 Timestamp { get; set; }
        public Double Par_Alt { get; set; }
        public Double Par_Neu { get; set; }
        public Int16 ProgrammNr { get; set; }
        #region 2 Byte = 16 Bools
        public Boolean Bool_Neu { get; set; }
        public Boolean Bool_Alt { get; set; }
        public Boolean Reserve_2 { get; set; }
        public Boolean Reserve_3 { get; set; }
        public Boolean Reserve_4 { get; set; }
        public Boolean Reserve_5 { get; set; }
        public Boolean Reserve_6 { get; set; }
        public Boolean Reserve_7 { get; set; }
        public Boolean Reserve_8 { get; set; }
        public Boolean Reserve_9 { get; set; }
        public Boolean Reserve_10 { get; set; }
        public Boolean Reserve_11 { get; set; }
        public Boolean Reserve_12 { get; set; }
        public Boolean Reserve_13 { get; set; }
        public Boolean Reserve_14 { get; set; }
        public Boolean Reserve_15 { get; set; }
        #endregion  
        public Int16 Par_Id { get; set; }
        #endregion
    }
```

```
    public class DB
    {
        /***************************************************************************************
         * Must not contain other public properties than fillable by PLC maintaining order !!! *
         * aka "public Type propertyname { get; set; }"                                        *
         ***************************************************************************************/

        #region Contraints
        /// <summary>
        /// Offset to first entry in Bytes
        /// </summary>
        public int Offset;

        /// <summary>
        /// Length of structure (entry) in Bytes
        /// </summary>
        public int StructLen;

        /// <summary>
        /// Max count of entries
        /// </summary>
        public int MaxLen;

        /// <summary>
        /// DB number
        /// </summary>
        public int DBNo;
        #endregion
    }
```

My DBs have a main structure followed by identical sections, like DB800 (data logger). Therefor you can set an offset to start reading sections.

The DB800 is defined in Step7 and contains multiple UDTs following this section:
[image]

Hope it helps a little.

closed 2018-06-28 by Mallowan