S7NetPlus/s7netplus #109: Still can't get rid of 'NullReferenceException' while class or structure reading.

Mallowan, 2018-04-19:
Hello everyone my issue is still not resolved. :)
While playing around, I'm getting  NullReferenceException 'Object reference not set to an instance of an object.' Despite of  created 1Byte size DB1 in Step 7 project or 1Byte structure.
Can't get what's wrong.

`
using System;
using System.Collections.Generic;
using System.ComponentModel;
using System.Data;
using System.Drawing;
using System.Linq;
using System.Text;
using System.Threading.Tasks;
using System.Windows.Forms;
using S7.Net;

namespace WindowsFormsApplication1
{
    public partial class Form1 : Form
    {
        Plc plc = null;

        public Form1()
        {
            InitializeComponent();

            Plc plc = new Plc(CpuType.S7300, "127.0.0.1", 0, 2);
            plc.Open();
            if (!plc.IsConnected)
            return;

        }

        public class DB1
        {
            public bool varBool0 { get; set; }
            public bool varBool1 { get; set; }
            public bool varBool2 { get; set; }
            public bool varBool3 { get; set; }
            public bool varBool4 { get; set; }
            public bool varBool5 { get; set; }
            public bool varBool6 { get; set; }
            public byte varByte7 { get; set; }

        }

        private void button1_Click(object sender, EventArgs e)
        {

            DB1 db1 = new DB1();
            plc.ReadClass(db1, 1);

        }

        public struct DB1_Structure1
        {
            public bool varBool0;
            public bool varBool1;
            public bool varBool2;
            public bool varBool3;
            public bool varBool4;
            public bool varBool5;
            public bool varBool6;
            public bool varBool7;
        }

        private void button2_Click(object sender, EventArgs e)
        {
           DB1_Structure1 db1_structure1 = (DB1_Structure1)plc.ReadStruct(typeof(DB1_Structure1), 1);
        }
    }
}

`

Mallowan, 2018-04-19:
``

rapha-dev, 2018-04-19:
@Mallowan 
You have defined "plc" twice. First in class and second in Form1(). The instance used in button1_Clock() is never initialized.

It should be `plc = new Plc(CpuType.S7300, "127.0.0.1", 0, 2);` if you want to use the class defined instance.

You also want to add some error handling, i.e. not connected to PLC.

Mallowan, 2018-04-19:
yes, this is working.
[image]

> You also want to add some error handling, i.e. not connected to PLC.

>

rapha-dev, 2018-04-19:
Keyword here is [Watch](https://docs.microsoft.com/en-us/visualstudio/debugger/watch-and-quickwatch-windows).

Mallowan, 2018-04-19:
> You have defined only 7 bools and 1 byte. Keep in mind 1 byte = 8 bits (bools).
> Also the last byte is not "visible" in DB1 (PLC), it means it is not usable by PLC program (at least not symbolic) but still exists.
> 
@rapha-dev  I'm sorry, but I've defined 1Byte or 8 Bools, as seen on the screenshot above (VarBool0...varBool7) and in the DB1 (Varbool0...VarBool7).

> Set a Watch on Variables
Separate thank.

Mallowan

rapha-dev, 2018-04-19:
Have a look at the last "bool" in `class DB1`, it is a byte instead.
Your `public struct DB1_Structure1` consists of 8 bools.

From your original post:
```
public class DB1
        {
            public bool varBool0 { get; set; }
            public bool varBool1 { get; set; }
            public bool varBool2 { get; set; }
            public bool varBool3 { get; set; }
            public bool varBool4 { get; set; }
            public bool varBool5 { get; set; }
            public bool varBool6 { get; set; }
            **public byte varByte7 { get; set; }**

        }
```
PLC DB1 looks good.

Mallowan, 2018-04-19:
yes, later I have noticed that TYPO and corrected.

Mallowan

mesta1, 2018-05-13:
Closing as solved (code error on constructor).

closed 2018-05-13 by mesta1