S7NetPlus/s7netplus #112: an't read Dint and int. Help!!!

udon02, 2018-04-24:
Hi,

I have already used the new version s7 library in the past and worked really fine.
Now we are going to connect to an s7-400 plc and with this library it connects just fine.
The problem occurs when i try to read from the plc.

The var location is : DB71.DBW56
C# code: object s = plc1.Read("DB71.DBW56");
And i get WrongVarFormat.

Can you please help me out.

Regards,

rapha-dev, 2018-04-24:
Hi @udon02,

it is hard to guess, Read() always throws an WrongVarFormat on any error.
You may debug into ReadBytesWithASingleRequest() and see if there is data in receive buffer.

udon02, 2018-04-25:
Hi @rapha-dev 

i have tried it but still can not read it. now i can not read data from S7-400 all data type, but i can write data.

Can you please help me out.

Regards,

rapha-dev, 2018-04-25:
Can you try following instead?
`object s = plc1.Read(DataType.DataBlock, 71, 56, VarType.Word, 1);`
It should read the same variable, but you will see any other Exception.

udon02, 2018-04-26:
I have try but can not read.

rapha-dev, 2018-04-26:
@udon02 
Same error?
Should be `ErrorCode.WriteData` or `ErrorCode.WrongNumberReceivedBytes` if there is a known read error.

udon02, 2018-04-26:
is show ErrorCode : ErrorCode.WrongNumberReceivedBytes

rapha-dev, 2018-04-26:
So the received data from PLC is incorrect or better invalid.
Are you sure the DB exists in PLC and has sufficient length?

udon02, 2018-04-27:
@rapha-dev 
I'm sure there is a DB in the PLC.

rapha-dev, 2018-07-06:
@udon02 
We have improved the s7netplus library significantly and fixed many bugs, can you please test current snapshot?

rapha-dev, 2018-07-07:
Closing this for the moment, feel free to re-open.

closed 2018-07-07 by rapha-dev