S7NetPlus/s7netplus #113: Cant read data from S7-1200

hakfre, 2018-04-24:
I am unable to read data from a datablock in a S7-1200 (6ES7 212-1BE40-0XB0) with firmware V4.2. The software connect to the PLC but reading data returns only error messages.

For instance:
`var io0 = plc.Read("DB1.DBX0.0");`
gives the error: "The variable'DB1.DBX0.0' could not be read. Please check the syntax and try again."
and
`byte[] ioByte = plc.ReadBytes(DataType.DataBlock, 1, 0, 1);`
returns "WrongNumberReceivedBytes"

I made some debugging and found the problem to be the PLC not responding to the request on the socket-communication as expected. On line 1051 to 1054 in plc.cs the following code throws the exception:
`byte[] bReceive = new byte[512];`
`int numReceived = _mSocket.Receive(bReceive, 512, SocketFlags.None);`
`if (bReceive[21] != 0xff)`
`throw new Exception(ErrorCode.WrongNumberReceivedBytes.ToString());`

the array bReceive contains some data, hence the PLC responds to the request from the PC, but byte 21 is zero.

And yes, I did enable "PUT/GET" communication with the PLC and disabled "optimized block access" for the datablock.

Any input on further debugging?

rapha-dev, 2018-04-24:
Hi @hakfre,

had the same issue when reading from a S7-1500. Be sure to enable PUT/GET and download hardware to PLC. Just ticking the checkbox is not enaugh.

hakfre, 2018-04-24:
Thanks!
"Download the hardware" did the trick. I assumed that the "Download to device - Software (all)" also update the hardware configuration, but no.

Actually the "Download to device - Hardware configuration" is only available when you are offline. As soon as the PLC is online in TIA the option dissapear, thats probably why I missed it.

closed 2018-04-24 by hakfre