S7NetPlus/s7netplus #116: When PLC is not reachable an inappropriate exception is thrown

mbalous, 2018-04-26:
Disconnecting the PLC from power supply after `plc.Open()` is invoked produces an exception specified below:
```
Could not read value from PLC. 
Exception ocurred when trying to read value from PLC. 
Exception error string: 'The variable'DB111.DBX1.2' could not be read. 
Please check the syntax and try again.'.
```

I think this is really misleading, because a syntax correction definitely won't help. An exception with message similar to "PLC can not be reached" should be thrown in my opinion.

Mallowan, 2018-04-26:
> An exception with message similar to "PLC can not be reached" should be thrown in my opinion.
Yes, you're right, provided that your application polls continuosly such lib properties like 'IsConnected'.
And based on this event your app disable a var polling.

Mallowan

rapha-dev, 2018-04-27:
Right, similar to issue #112 where WrongVarFormat is returned.
Methods `Read(string)` and `Write(string, object)` should be affected. Needs better exception handling (Exception forwarding).

rapha-dev, 2018-07-07:
Addressed in #156.

closed 2018-07-07 by rapha-dev