S7NetPlus/s7netplus #120: Problem reading struct

jonweaver, 2018-05-16:
I working on a project with an S71200 PLC and it has a DB with 3 words and 16 booleans.  My struct is below.  When I try to read this DB from the PLC using

`   var data = (FromPLCStruct)_plc.ReadStruct(typeof(FromPLCStruct), 8);`

I get an IndexOutOfRange error.

I tracked it down to here.  It appears that COTP.TSDU.Read is returning a byte array with only 12 bytes. I'm not sure what it should be sending but it is obviously supposed to be more than 12.  Is this maybe an error code or something?

[image]

` public struct FromPLCStruct
    { 
        public ushort BkStatus1;
        public ushort BkStatus2;
        public ushort BkStatus3;
        public bool HeartbeatOk;
        public bool Spare1;
        public bool Spare2;
        public bool Spare3;
        public bool Spare4;
        public bool Spare5;
        public bool Spare6;
        public bool Spare7;
        public bool Spare8;
        public bool Spare9;
        public bool Spare10;
        public bool Spare11;
        public bool Spare12;
        public bool Spare13;
        public bool Spare14;
        public bool Spare15;
    }`

mesta1, 2018-05-16:
@thoj can you have a look ?

jonweaver, 2018-05-17:
I found the problem.  The PLC programmer forgot to check the "Permit access with Get/Put..." on the PLC.  If this is an error code it should be possible to pass it right?

closed 2021-06-06 by mycroes