S7NetPlus/s7netplus #122: What exactly means second number in the reading DB class method?

Mallowan, 2018-05-28:
I'm confused. While testing I didn't use numbers of DBs large than "2" and everything was working fine.
For example:

```
DB1 db1 = new DB1();
plc.ReadClass(db1, 1); // reading DB1

DB2 db2 = new DB2();
plc.ReadClass(db2, 2); // reading DB2
```
 I tried to read DB called DB9 and I've got exception.
[image]

So I would like to know what really means second number in the reading method. Should there be '9'?

rapha-dev, 2018-05-29:
Hi,

your ReadClass signature would lead to `public int ReadClass(object sourceClass, int db, int startByteAdr = 0)`, where startByteAdr is optional and defaults to 0.

So `plc.ReadClass(db9, 9)` is correct if you want to read from DB9 starting at byte 0.

Mallowan, 2018-05-29:
@Rapha-dev,
I'm sorry. That's amazing TIA portal...  As they say, 'The morning is wiser than the evening'. Sometimes TIA requires whether a reboot of OS, whether a reboot of TIA itself. I didn't that catch. Just I had run my laptop today morning and yesterday's code has became working... ) Awesom stuff...

mycroes, 2018-05-30:
@Mallowan Could you close the issue please? No need to keep it open.

closed 2018-05-30 by Mallowan