S7NetPlus/s7netplus #128: v0.1.9 PLC ,The remote host forced the closure of an existing connection.

flyasme, 2018-06-21:
When I use the v0.1.9 PLC , I received the message of "The remote host forced the closure of an existing connection." But if I use the v0.1.8, the Open is OK.
I want to know how to fix it if I use v0.1.9.

mycroes, 2018-06-21:
What type of PLC are you connecting to? Can you share a minimal reproduction?

flyasme, 2018-06-22:
S7-400. 416-3  and cp443-1
if(plc.IsAvailable=true){
plc.open();
if(plc.IsConnected){ 
....//Here received the error message!
}
}

mycroes, 2018-06-22:
Can you post your `new Plc(...)` line as well? I'll compare if something has changed with regards to opening communication. Also, you're only referring to `plc.IsConnected` with regards to error messages, but `plc.IsConnected` never throws. What is the actual call that causes an error / throws an exception? What is the message?

mycroes, 2018-06-22:
@flyasme could you check out the commit that I added to [mycroes/s7netplus](https://github.com/mycroes/s7netplus/tree/use-256-byte-pdu-in-communication-setup) and see if that works? Recently the requested PDU size on the client side (s7netplus) was changed from *256* to *1872* (the 1872 is an error, should have been *1920*, see PR #130 for details). I can imagine S7-400 might bail on this, but I'm not sure.

Also, if you can provide accurate information as to the exception you're getting (on what s7netplus call do you get it? What is the message? Does it contain a filename and linenumber?) we will probably have this fixed in no time.

flyasme, 2018-06-23:
Code:
 fur1PLC = new S7.Net.Plc(par.CPU, "192.168.0.3", par.rack, par.slot);
 rack=0	slot=3
 
 fur1PLC.Open(), cause an error, the message is "The remote host forced the closure of an existing connection."
 The errorCode is "ConnectionError".
 
 About PDU , I'll try it the next workday.

andrea5586, 2018-06-26:
In my case with S7300 with CP343-1 Lean the call PLC.Open() gives errorCode= "connectionError" and CP goes system fault.

mycroes, 2018-06-26:
I actually verified the local and destination TSAP for all CPU types yesterday, so my guess is still on PDU size (it's one of the few things that has actually changed). @andrea5586 could you check if https://github.com/mycroes/s7netplus/tree/use-256-byte-pdu-in-communication-setup works for you with S7-300?

andrea5586, 2018-06-28:
@mycroes Thank you for your suggestion. At the moment i don't have the S7300 available. I will try your solution when it will be reachable again.

bauerju, 2018-07-03:
Hi I also had similar problems with connections:

Before reading or writing to plc (S7 319-3 PN/DP) checked plc.IsAvailable. Which I later found that every call openes a new TCP-connection to the plc. This overflowed the connection limit of the plc. (You could see this in Sysinternals tcpview : Every time when IsAvailable was called a new tcp-Connection was opened). I also got massive System.IO and TPKTInvalidException exceptions. 
This happend only with the actual version from github. 
With the Nuget package 0.1.9  didn't have this feature. 
Looking at the source IsAvailable has also an "TODO tag.

After removing this call. Everything worked fine. May bee this Information could help you. 
Jürgen

bauerju, 2018-07-03:
Hi @rapha-dev 
Thank you for your reply. I had switched to `IsConnected`. No more faults. I think you should re-use the socket as long it is not closed by the plc.  

Jürgen

mycroes, 2018-07-03:
@andrea5586 @flyasme Did any of you get to testing this yet? I also opened an issue looking for testers for S7-400 (issue #148), would be very helpful if you guys would be able to test changes (now and perhaps in future as well).

andrea5586, 2018-07-06:
Hi @mycroes!
I tested your suggestion and it works! Here the snippet I used:
```
S7.Net.Plc plc = new S7.Net.Plc(S7.Net.CpuType.S7300, "192.168.27.135", 0, 2);
plc.Open();
var bytes = plc.ReadBytes(S7.Net.DataType.DataBlock, 202, 0, 10);
```
No connection error and no system fault on CP343!

mycroes, 2018-07-06:
@andrea5586 Could you also test current state of kllnine/s7netplus to see if it works here as well? The negotiated PDU size was not a power of 2, so either that breaks it or the fact that it's too large causes issues for the S7-400.

andrea5586, 2018-07-06:
The same snippet tested with current commit (db50a62) of killnine/s7netplus works properly!

mycroes, 2018-07-06:
@andrea5586 Thanks for testing! @flyasme Can you also check the current state of killnine/s7netplus to see if it works for you as well?

mycroes, 2018-07-07:
I'm assuming this is fixed now (although we can't be sure without feedback from @flyasme), please re-open if the issue persists.

closed 2018-07-07 by mycroes

noumanqaiser, 2019-12-17:
Hi team, I have verified and was successful in reading through S7 300 via a CP343-1 with the latest nuget package. The error "The remote host forced the closure of an existing connection." now appears only if you use an invalid slot number(1), I used slot number 2 while defining the s7 300 PLC and it worked successfully.

Regards,
Nouman

flyasme, 2019-12-17:
Hello NoumanFirst thankyou very much !I will try it latter with your method. The project is running very smoothly and stably. 发自我的华为手机-------- 原始邮件 --------发件人： noumanqaiser <notifications@github.com>日期： 2019年12月17日周二 下午4:23收件人： S7NetPlus/s7netplus <s7netplus@noreply.github.com>抄送： flyasme <gaoag@189.cn>, Mention <mention@noreply.github.com>主    题： Re: [S7NetPlus/s7netplus] v0.1.9 PLC ,The remote host forced the closure of an existing connection. (#128)Hi team, I have verified and was successful in reading through S7 300 via a CP343-1 with the latest nuget package. The error "The remote host forced the closure of an existing connection." now appears only if you use an invalid slot number(1), I used slot number 2 while defining the s7 300 PLC and it worked successfully.
Regards,
Nouman