S7NetPlus/s7netplus #131: struct and class read error , when read real numbers.

flyasme, 2018-06-22:
If I read one real number one a time , it's ok.
But if I used ReadClass or ReadStruct ,I get some wrong numbers.
And bool type and int type is right.

flyasme, 2018-06-22:
Do you have any suggestion about it?

flyasme, 2018-06-22:
It's ok, changed the sequence of the vars in  DB, all real vars must in the head of the DB, then change to the same of the class.

mycroes, 2018-06-22:
Bit reading has had some issues, I'm guessing it still doesn't work properly in ReadClass / ReadStruct.

mycroes, 2018-06-22:
@flyasme I think you're using the method incorrectly. I'm assuming you have the following struct:
```csharp
struct RD
{
    public bool varBool1;
    public bool varBool7;
    public bool varBool8;
    public double varReal1;
    public double varReal2;
    public double varReal3;
    public double varReal4;
}
```
_(BTW s7netplus inccorectly translates Real to double, the matching .NET type for a 4-byte floating point number is float / Single)._

The signature is as follows: `public object ReadStruct(Type structType, int db, int startByteAdr = 0)`. What this means is that your call `ReadStruct(typeof(RD), 20)` will attempt to read the struct from **DB20**, starting at byte **0**. While byte 0 does contain the bits you want to read, the call to `ReadStruct` (assuming the struct as defined) will read
- **DB20,0.0**
- **DB20,0.1**
- **DB20,0.2**
- **DB20,R2**
- **DB20,R6**
- **DB20,R10**
- **DB20,R14**

Seeing that your floats are at **DB20,R44**, ... it makes sense that you get different data.

flyasme, 2018-06-23:
First of all , thank you !
I will test more, and feedback here for peers.
And I should to say sorry ,for  I didn't post the detail of my code.
The last version is ...,I test word ,bit and real ,with this code, and Received the same value.
    public  class ToRead
    {
        public double DB_VAR2215 { get; set; }  //Here I've changed yesterday.
        public double DB_VAR2216 { get; set; }// The original is public double DB_varReal1{get;set}
        public double DB_VAR2221 { get; set; }
        public double DB_VAR2231 { get; set; }
        public double DB_VAR2241 { get; set; }
        public double DB_VAR2251 { get; set; }
        public double DB_VAR2261 { get; set; }
        public double ToL2_LadleTemp { get; set; }
        public double ToL2_GasPress { get; set; }
        public double ToL2_GasFlux { get; set; }
        public double ToL2_AirFlux { get; set; }
        public double ToL2_SmokeFlux { get; set; }
        public double ToL2_GasValFB { get; set; }
        public double ToL2_AirValFB { get; set; }
        public double L222 { get; set; }
        public double DB_VAR228 { get; set; }
        public double DB_VAR229 { get; set; }
        public double DB_VAR2210 { get; set; }

        public bool ToL2_FlameSignal { get; set; }
        public bool ToL2_FireGas { get; set; }
        public bool ToL2_GasValveOpn { get; set; }
        public bool ToL2_GasValveCls { get; set; }
        public bool ToL2_NitrogenBlow { get; set; }
        public bool ToL2_LadleTempLost { get; set; }
        public bool ToL2_ValReversing { get; set; }
        public bool ToL2_DraughtFanStart { get; set; }
        public bool ToL2_SysFault { get; set; }
        public bool ToL2_DraughtFanStop { get; set; }
        public bool varBool11 { get; set; }
        public bool varBool12 { get; set; }
        public bool varBool13 { get; set; }
        public bool varBool14 { get; set; }
        public bool varBool15 { get; set; }
        public bool varBool16 { get; set; }
        public bool ToL2_AutoStart { get; set; }
        public bool ToL2_FaultReset { get; set; }
        public bool varBool19 { get; set; }
        public bool varBool20 { get; set; }
        public bool varBool21 { get; set; }
        public bool varBool22 { get; set; }
        public bool varBool23 { get; set; }
        public bool varBool24 { get; set; }
        public bool varBool25 { get; set; }
        public bool varBool26 { get; set; }
        public bool varBool27 { get; set; }
        public bool varBool28 { get; set; }
        public bool varBool29 { get; set; }
        public bool varBool30 { get; set; }
        public bool varBool31 { get; set; }
        public bool varBool32 { get; set; }
        public bool varBool33 { get; set; }
        public bool L2_Stop { get; set; }
        public bool L2_AutoStart { get; set; }
        public bool L2_FaultReset { get; set; }
        public bool L2_Fire { get; set; }
        public bool varBool38 { get; set; }
        public bool varBool39 { get; set; }
        public bool varBool40 { get; set; }
        public bool L2_ControlMode { get; set; }
        public bool L2_NitrogenBlow { get; set; }
        public bool varBool43 { get; set; }
        public bool varBool44 { get; set; }
        public bool varBool45 { get; set; }
        public bool varBool46 { get; set; }
        public bool varBool47 { get; set; }
        public bool varBool48 { get; set; }

        public ushort WatchDog_L2 { get; set; }
        public ushort varWord2 { get; set; }
        public ushort varWord3 { get; set; }
        public ushort varWord4 { get; set; }
        public ushort varWord5 { get; set; }
        public ushort varWord6 { get; set; }
        public ushort varWord7 { get; set; }
        public ushort varWord8 { get; set; }
        public ushort varWord9 { get; set; }
        public ushort varWord10 { get; set; }
        public ushort varWord11 { get; set; }
        public ushort varWord12 { get; set; }
        public ushort varWord13 { get; set; }
        public ushort varWord14 { get; set; }
        public ushort varWord15 { get; set; }
        public ushort varWord16 { get; set; }
        public ushort varWord17 { get; set; }
        public ushort varWord18 { get; set; }   

    }

mycroes, 2018-06-26:
I'm sorry for my late reply, but it seems your comment doesn't make sense. 

> ```csharp
> public class ToRead
> {
>     public double DB_VAR2215 { get; set; } //Here I've changed yesterday.
>     public double DB_VAR2216 { get; set; }// The original is public double DB_varReal1{get;set}
>     ....
> }
> ```

The comment says you swapped a `double` for a `double`, that doesn't make sense. If you'll post the original struct or class I'll see if there's something wrong with the calculation.

Also, it's a bad idea to read a class this way, because order of `GetProperties()` is not guaranteed to match anything. That's not your fault, it's a library defect (which is now known at #135). Possibly that could be the cause of your issue, but if you can at least provide the requested data I can check if there are other issues.

closed 2021-06-06 by mycroes