S7NetPlus/s7netplus #132: s7netplus incorrectly uses `double` as the .NET type for REAL values in PLC

mycroes, 2018-06-22:
`Double` values in .NET are 8 bytes long, `float` / `Single` are 4 bytes as is the `REAL` type in Siemens PLC's. Also, except for byte ordering the bitwise values are equal.

@mesta1 My proposal would be to mark all `double` related conversions as obsolete and replace them with `float` conversions. I'm not sure what would be sane way to handle double fields in structs/classes, my personal opinion would be to throw an exception on those and up the s7netplus version to 2.0.0.

rapha-dev, 2018-06-23:
I'd like to vote +1 on this as I had some kind of misunderstanding before.

rapha-dev, 2018-07-02:
Have prepared the implementation of float / Single, needs some testing against real hardware but should work. Will also create a PR for this.

mycroes, 2018-07-03:
Fixed by #143.

closed 2018-07-03 by mycroes