S7NetPlus/s7netplus #135: ReadClass and ReadStruct order is not guaranteed

mycroes, 2018-06-26:
Title says it all. Marc Gravell has a good answer [over on StackOverflow](https://stackoverflow.com/questions/5473455/c-sharp-get-fieldinfos-propertyinfos-in-the-original-order/5473526#5473526) as how this can be solved. For structs there's an actual solution that's also described at dotnet/corefx#14606; if `LayoutKind` is applied on a struct the `MetadataToken` property of the `FieldInfo` can be used to order the fields. I'm not sure if this even applies to classes, but if it does that's at best an _undocumented implementation detail_.

I think it would be a good idea to check if `LayoutKindAttribute` is defined on structs (and apply the `MetadataToken` ordering), as far as classes are concerned I guess a custom attribute would be the way to go.

rapha-dev, 2018-06-29:
That is a good idea, indeed, didn't knew that. So it should be possible to use LayoutKind to get the properties in order.
Another approach would be to explicitly order the properties to compile time as seen [here](https://stackoverflow.com/a/17998371/6874271)?

mycroes, 2018-06-29:
That's a good find, very neat trick. As far as `struct`s are concerned, just specifying LayoutKind is enough to guarantee field order by MetadataToken. I'm not sure what we want to force on people, but creating an S7FieldAttribute which includes the linenumber trick could be nice. That would also permit overrides or specifying string length for instance.

I was actually thinking if it perhaps would be a good idea to have this logic in a separate library. I didn't really care for implementing class / struct reading in Sally7, but if we can implement it in a standalone library (that would work for both Sally7 and S7NetPlus and possibly sharp7, without actually depending on them) I can see some advantages.

ccirstea, 2018-10-10:
Hi guys,

     I seem to be facing this issue with the ReadClass(..) method that you have mentioned. 
     In my c# program I read 19 registers using ReadClass(..). They are not complicated and do not contain may variables, mostly several bools each (as they are directly linked to sensor outputs). 
     I think it may be ugly to use Read() for each var of the register as this implies multiple calls to Read(), which may incur overhead, and since using a struct in stead of class doesn`t solve the problem i was wondering what do you recommend as a solid solution?

Thank you,

mycroes, 2018-10-10:
@ccirstea Are you actually having issues with the order of items read? There could be issues with offset calculation, but my main concern is you can't easily see the calculated offsets either... I intend to fix that at some point, not sure when though.

ccirstea, 2018-10-10:
@mycroes I am having issues with a boolean directly connected to a break the beam sensor. The Db class looks like this: 

    [Location(32)]
    public class PlcRevolverStatusDb
    {
        public bool Initialized { get; set; }
        public bool InputSlotAbsenceState { get; set; }
        public bool OutputSlotAbsenceState { get; set; }
        public bool ProcessingAreaAbsenceState { get; set; }
        public ushort DummyValue { get; set; }
    }

ProcessingAreaAbsenceState directly connected to the sensor sometimes reads opposite to the presence. Could the ushort influence this faulty read?

mycroes, 2018-10-10:
@ccirstea Moved to #195, please add the code you use for reading the class there.

scamille, 2020-08-25:
https://stackoverflow.com/questions/8067493/if-getfields-doesnt-guarantee-order-how-does-layoutkind-sequential-work has some more/same discussion about this.

There is also a link in there pointing to https://github.com/dotnet/runtime/issues/19732 which confirms that LayoutKind.Sequential is required.

scamille, 2020-08-25:
I started a PR for this. The main question is if it should enforce the struct types to have [StructLayout(LayoutKind.Sequential)] or not. I can't find any statements saying that the ordering is messed up in practice, even though in theory the compiler/runtime could reorder things.

Question is: Is there any evidence that this ever caused a problem at all, or is it purely a hypothetical problem?
(And if there was a problem, an example Struct would be great so this can be checked with a unit test)