S7NetPlus/s7netplus #137: Exception vs LastErrorCode / LastErrorString

mycroes, 2018-06-26:
As the title says, use of `Exception`s vs `LastErrorCode` and `LastErrorString`. Personally I favor exceptions because they don't break under concurrent use and they're the builtin (as in within the language) solution for this problem. A custom exception type can be created to also return the error code. There are some cases where exceptions are thrown on precondition failures while in other cases the error code and message are set, I think this should be implemented consistently.

rapha-dev, 2018-06-28:
Agreed. Some methods are also producing misleading error strings, this should be cleaned up while implementing Exceptions.

rapha-dev, 2018-07-11:
Merged into `develop`.

closed 2018-07-11 by rapha-dev