S7NetPlus/s7netplus #139: Writing a DInt to DB. I got stuck on this matter.

Mallowan, 2018-06-29:
Hello,
Trying to write a DInt, followed provided code samples and Documentation.
I've got an exception as shown below.
[image]
and from TIA DB's graph.
[image]
Could anyone give me some advices?
Thanks in advance!

Mallowan

mycroes, 2018-06-29:
Please provide the StackTrace of the Exception so we can trace the origin.

Mallowan, 2018-06-29:
`   at S7net_01.Par_Mixer.Par_Mixer_FormClosed(Object sender, FormClosedEventArgs e) in C:\Users\NGC_user\Documents\Visual Studio 2015\Projects\S7net_01\S7net_01\Parameter_Form\Par_Mixer.cs:line 99
   at System.Windows.Forms.Form.OnFormClosed(FormClosedEventArgs e)
   at System.Windows.Forms.Form.WmClose(Message& m)
   at System.Windows.Forms.Form.WndProc(Message& m)
   at S7net_01.Custom_Forms.BorderLessForm.WndProc(Message& m) in C:\Users\NGC_user\Documents\Visual Studio 2015\Projects\S7net_01\S7net_01\Custom_Forms\BorderlessForm.cs:line 156
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.DebuggableCallback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
   at System.Windows.Forms.UnsafeNativeMethods.SendMessage(HandleRef hWnd, Int32 msg, Int32 wParam, Int32 lParam)
   at System.Windows.Forms.Control.SendMessage(Int32 msg, Int32 wparam, Int32 lparam)
   at System.Windows.Forms.Form.Close()
   at S7net_01.MainForm.button1_MouseDown(Object sender, MouseEventArgs e) in C:\Users\NGC_user\Documents\Visual Studio 2015\Projects\S7net_01\S7net_01\MainForm.cs:line 392
   at System.Windows.Forms.Control.OnMouseDown(MouseEventArgs e)
   at System.Windows.Forms.ButtonBase.OnMouseDown(MouseEventArgs mevent)
   at System.Windows.Forms.Control.WmMouseDown(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ButtonBase.WndProc(Message& m)
   at System.Windows.Forms.Button.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.DebuggableCallback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
   at System.Windows.Forms.UnsafeNativeMethods.DispatchMessageW(MSG& msg)
   at System.Windows.Forms.Application.ComponentManager.System.Windows.Forms.UnsafeNativeMethods.IMsoComponentManager.FPushMessageLoop(IntPtr dwComponentID, Int32 reason, Int32 pvLoopData)
   at System.Windows.Forms.Application.ThreadContext.RunMessageLoopInner(Int32 reason, ApplicationContext context)
   at System.Windows.Forms.Application.ThreadContext.RunMessageLoop(Int32 reason, ApplicationContext context)
   at System.Windows.Forms.Application.Run(Form mainForm)
   at S7net_01.Program.Main() in C:\Users\NGC_user\Documents\Visual Studio 2015\Projects\S7net_01\S7net_01\Program.cs:line 19
   at System.AppDomain._nExecuteAssembly(RuntimeAssembly assembly, String[] args)
   at System.AppDomain.ExecuteAssembly(String assemblyFile, Evidence assemblySecurity, String[] args)
   at Microsoft.VisualStudio.HostingProcess.HostProc.RunUsersAssembly()
   at System.Threading.ThreadHelper.ThreadStart_Context(Object state)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.ThreadHelper.ThreadStart()`

Mallowan, 2018-06-29:
I think this is my mistake. But where ithis is? :)

rapha-dev, 2018-06-29:
Hi @Mallowan,
you have a syntax error in your variable. `DB8.DBD2.0` is not valid, you may use `DB8.DBD2` for DINT.

Mallowan, 2018-06-29:
Hi @rapha-dev, long time, no see! )
DB8.DBD2.0 this is normal absolute addresing for Step7 and for TIA Portal, if it is turned on to absolute addresing in its settings.
Perhaps I did it reflexively. there's a lot of work for a single brain. the heap of hardware, two apps, different prog. languages and so one... )

Mallowan
P.S. just checked your vesrsion. I've got same exception.

rapha-dev, 2018-06-29:
I'm pretty sure it is a syntax error in Step7, unsure about TIA.
Can you post exception details please?

rapha-dev, 2018-06-29:
Yeah, dealing with multiple projects at the same time can be exhausting.

The first (essential) part of your exception detail is missing, it should be written before `at S7net_01.Par_Mixer.Par_Mixer_FormClosed...`.

Mallowan, 2018-06-29:
espesially considering that some part of brain is busy with FIFA World Cup  games. ) I will look...

rapha-dev, 2018-06-29:
Right, you posted the StackTrace, important part here is `NullreferenceException`. StackTrace shows where it was thrown, in your case in the `FormClosed` method.

Same code works when you replace DB8 with DB9?

Mallowan, 2018-06-29:
it seems, I've found  where the issue is.
Just created button_click event handler at MainForm and sent to DB8 a value. That went well. DB8 got a value. 
However  writing doesn't work from events that are situated at subForm. ( There's no syntaxis fails there, though.

Mallowan, 2018-06-29:
Where I can see **plc** instance `NullReferenceException`?

rapha-dev, 2018-06-29:
That is something common, when you create an instance of a class (or anything else) and it isn't static you cannot access the instance from outside.

Mallowan, 2018-06-29:
It's hard to understand...
It would be nice to look at a sample for second form.

rapha-dev, 2018-06-29:
Maybe [this](https://stackoverflow.com/a/4660186/6874271) helps to understand why you get that `NullReferenceException`, will try to get an example.

mycroes, 2018-06-29:
Tagging this as invalid because it doesn't seem related to S7NetPlus in any way. I'll leave it open for now because I guess you're almost there, but I'd rather not have totally unrelated issues here.

rapha-dev, 2018-06-29:
Sure, better to discuss in another repo (see above).

Mallowan, 2018-06-29:
@mycroes, Hi there,
Put yourself in my shoes and try to write on (e.g. Stackoverflow) like: _'I've got an exception while running HMI based on S7netPlus library and don't know whether this exep comes from used library itself or I made some mistakes in C# code because I'm newbee, or it's double issue.' Like it really have been.
I guess most of site members will make big eyes on this request.
Mallowan.

@rapha-dev , example, this is great stuff, THANK YOU A LOT. Finally wrote to DB from SubForm. )

`if (MainFormsPlc != null)
            {
                var value = MainFormsPlc.Read("DB9.DBD2");
            }`
tell me please for what, I'm asking because I have in my code following:
`private Plc plc = null;`

Thanks in advance!

rapha-dev, 2018-06-29:
It is an example for reading in a SubForm. `var value` can be anything else, it just should demonstrate how to use the passed Plc instance without creating a new instance in every Form.

Do you mind if we discuss in my example repo as it really doesn't belong here? https://github.com/rapha-dev/examples/issues

mycroes, 2018-06-29:
Dear @Mallowan (sincerely with best intents, no sarcasm),

> Put yourself in my shoes [...]

I did, that's why I didn't close the issue. I've labeled it so that if someone is searching for actual issues to fix (like me, since I'm trying to get S7NetPlus into better shape) they can skip issues that are not really related to S7NetPlus. There is no offense meant by labeling the issue 'invalid' and I think you should be very thankful for @rapha-dev for the time he spent on helping you. Also, you should try to avoid taking things personally (whenever possible of course) when people offer criticism that can help you improve your code (or code related skills). _What's the chance you actually considered putting yourself in my shoes?_

> [...] (e.g. Stackoverflow) like [...]

StackOverflow has some [clear guidelines](https://stackoverflow.com/help/how-to-ask). A very tiny excerpt:
> Include just enough code to allow others to reproduce the problem. For help with this, read How to create a Minimal, Complete, and Verifiable example.

If you would've taken such advice, you would've concluded that you didn't hit an issue in S7NetPlus. We're not here to support beginning programmers. That doesn't mean we don't want to help you, but you can probably get way better support in a place that is meant to support beginning programmers. One such place that I can recommend is [codingame.com](https://www.codingame.com/). It is in no way related to S7NetPlus, but it offers a nice way to learn coding (although it's not a tool to learn coding per se, it offers coding challenges of different levels of complexity).

Now back to your issue (because I don't mind to help you, I just want to keep the issues clean): maybe you should try to shrink your problem domain. You're having multiple forms, you're trying to share stuff and my guess is you're new to programming and not just new to C# (again, no offense meant, just guessing / stating). Try to cut stuff into pieces. Create a sample application that reads from PLC but doesn't involve any UI (so create a command line program). Create a sample program that does involve a UI and tackles a different part of your project. If I want to test something I often create a small project which I name [feature]Sandbox, here's my current list:
- AndroidSandbox/
- AspNetCoreSandbox/
- AspNetMvcSandbox/
- BsonSandbox/
- CaliburnMicroSandbox/
- ContinueSandbox/
- DelegateDecompilerSandbox/
- DynamicDataSandbox/
- EFSandbox/
- EventSandbox/
- ExpressionSandbox/
- FormsSandbox/
- HttpClientSandbox/
- IlSandbox/
- PagedListJsonSandbox/
- ReactiveUISandbox/
- ReflectionSandbox/
- RegisterBuilderSandbox/
- RestSandbox/
- Sally7Sandbox/
- SerializationSandbox/
- SystemMemorySandbox/
- TranslationSandbox/
- WebSandbox/
- WpfSandbox/
- XmlRpcSandbox/
- YieldSandbox/

Besides sandboxes I also use the **C# Interactive window** (in Visual Studio) a lot, just Google to see how you can make use of it.

Now even more closely related to your issue. I'd be happy to take a look at your code, but S7NetPlus is not the place where it should end up. If you can create your own repository I can help you there. When you have a repository I can easily fetch your code to reproduce any issues and I can provide Pull Requests to fix them (if you so desire).

Last but not least, S7NetPlus is missing a sample program. A project like S7NetPlus should be a community effort (although it seems it has mostly been a single maintainer at a time). A community is not just made up of (good) developers. A community needs users and as a project S7NetPlus could use contributions from those users as well. So maybe you could actually provide a sample implementation that we can include in the repository for future reference for new users. I'd be happy to assist in creating such a sample, if you don't feel like contributing it I will probably end up doing it myself anyway just as well...

To summarize, I hope I have given you some pointers that can assist you in your learning challenge. If you want help with your project, I'll be happy to help but I do expect that you create a repository for your program in that case. Last but not least I welcome you to contribute to the S7NetPlus project.