S7NetPlus/s7netplus #140: Changelog / Releases out of sync

rapha-dev, 2018-06-29:
We have [Changelog](https://github.com/killnine/s7netplus/wiki/Changelog) in the Wiki and [Releases](https://github.com/killnine/s7netplus/releases) in Code.
Maybe Changelog in Wiki should be removed and Releases (including description) should be used instead as they're not matching already.

mycroes, 2018-06-29:
If we properly add release information to the version tags the releases will contain that information (see [Sally7/releases](https://github.com/mycroes/Sally7/releases) for a sample). I think we *could* add more information to releases, but a decent tag annotation *will* do as release info.

About the Changelog in the Wiki, I propose we delete that. I guess everyone that cares about the changelog would rather take the changes from the commit history.

rapha-dev, 2018-06-29:
Tag annotations should be sufficient, it can be more info on major releases to give a brief intro to what essentially changed. I'm looking forward to major changes like #135 as it _could_ break existing implementations (not that it should).
I have seen release 0.1.9 out in NuGet, but Tag is not existing in repo. Can it be added afterwards? (Still learning to handle GitHub)

mycroes, 2018-06-29:
Tags, GitHub and NuGet are all kinda separate. Although GitHub could theoretically limit creation of tasks, that just wouldn't make sense. As far as NuGet is concerned, nothing promises that the NuGet package is actually built from the source in the repository.

For Sally7 I'm actually using AppVeyor to deploy to NuGet. That way I'm sure that the NuGet package matches the sources in git and it's actually pretty easy as well.

I'll add the 0.1.9 tag, will see what was changed since 0.1.8.

mycroes, 2018-06-29:
0.1.9 is in releases now.

mycroes, 2018-07-01:
I deleted the Changelog page and since 0.1.9 is releases now I'll close this.

closed 2018-07-01 by mycroes