S7NetPlus/s7netplus #145: ReadBytesWithSingleRequest cannot read >491 Bytes on S7-1500

rapha-dev, 2018-07-03:
Possible bug in COTP CR when connecting to a S7-1500, TPDU is too small (actually 512 bytes). Any Read resulting in `actualPDU + COTPHeader > TPDU` (which is 960 + 3 = 963 > 512) will fail (Socket disconnects).

Maximum Bytes of 491 would work: 509 + 3 = 512 > 512

For reference: [PLC4X](https://plc4x.incubator.apache.org/protocols/s7/s7comm.html)

Increasing the TPDU to maximum of 1024 bytes (still need to check on S7-300, 400 and 1200) will allow to read an unspecified maximum of bytes as PDUs will be split into multiple PDUs with maximum size.

I will create a PR for this and need some testing.

mycroes, 2018-07-03:
This probably also relates to open issue for PR #133.

> Increasing the TPDU to maximum of 1024 bytes

We're negotiating 1920 byte PDU, so maybe we should increase to 2048. Also, the _ISO over TCP_ spec states the default is **65355** bytes IIRC, which means we could possibly zero out the PDU size entirely. We should really have access to all supported PLC's to verify these changes though.

rapha-dev, 2018-07-03:
You are right, I have made some tests using different sizes up to 2^15, using 2^11 should be sufficient to support max PDU on all available PLCs.

rapha-dev, 2018-07-11:
Merged into `develop`.

closed 2018-07-11 by rapha-dev