S7NetPlus/s7netplus #147: Function to create a S7.Net.Types.DataItem from string

bauerju, 2018-07-03:
Hi 
As reading for multiple items can easy bee done with method  `ReadMultipleVars` which accepts a `List<S7.Net.Types.DataItem>` , a function to create a `S7.Net.Types.DataItem` from a string would be nice. 
Eventualy a constructor with string as input could be added. A SetValue which convertes into the used datatype would also help a lot. A new function to `Write( S7.Net.Types.DataItem value)` would also be nice. 
 
Jürgen

mycroes, 2018-07-03:
The Write function is already there, it takes `params DataItem[] dataItems` as argument, meaning you can pass in one, multiple or an array of `DataItem`s.

I'll take a look at adding the string addressing support, all the logic is already there.

bauerju, 2018-07-03:
Hi @mycroes 

Thanks, I was blind ;-) Didn't see it.  Does it also have the 20 Items / 222 byte length problem like the 
`ReadMultipleVars` method?

Jürgen

bauerju, 2018-07-03:
Hi @mycroes 
Ok. I'll do the same with the `ReadMultipleVars` : I sort the list and split it in several jobs.
Jürgen.
By the way : what is about strings or byte arrays? We use strings in our plc's (barcode from scanners) 
I didn't see a way to read it when I use the `Read(string)` method.  There is no way to decode the string as an Array of sth. nor as an string.
eg: 
DB100.DBB2[10] -> Array of 10 bytes, starting at byte 2

For strings the syntax could be like this:
DB100.DBS2[10] -> string with 10 chars starting at byte 2
DB100.DBB2[10]:STRING -> string with 10 chars starting at byte 2
DB100.DBB2[10]:STRINGEX -> string with 10 chars starting at byte 4 with Byte 2 and 3 are the length information: Byte 2: Maxmimum length, byte 3 used length.

rapha-dev, 2018-07-03:
Actually I have used something like [this snippet](https://gist.github.com/rapha-dev/f3cd66f69e6d141d2842008335db912c#file-readstring-snippet-cs) some time ago. It is not well teste though.

mycroes, 2018-07-03:
@bauerju @rapha-dev See [Types/StringEx](https://github.com/killnine/s7netplus/blob/master/S7.Net/Types/StringEx.cs).

Also, try to keep issues clean, if further discussion is warranted move it to a new issue so every issue contains a single problem and can be closed by a single fix.

bauerju, 2018-07-03:
Hi @mycroes @rapha-dev 
Thanks for your help. I'll open a new issue next time.  
Thank you.

Jürgen

bauerju, 2018-07-09:
Hi @mycroes 
I tested your added function from commit ecf2052e61a40ce83bda8b92b816f14de9d250b9.

Looks great. Thank you very much. Good Work.

But it lacks the support of arrays and setting other vartypes.
I can create a byte, word, doubleword from string but can't create an int, real (single),  string nor stringEX.
If you could add this in PlcAdress, this would also help when someone Reads or Writes from string definition.

Jürgen

bauerju, 2018-07-10:
Hit @mycroes 
May bee I tell you first for what I need this library at my work: 
I wrote a flexible windows service which first connects to a database to read all customizing like plc address and used plc tags. after initialization of the plc connections every tags gets and speaking name (`DB3100.DBX0.1` is eg `LiveBit` ). Some of them get polled (this is controlled in the customizing). Ever change triggers a sequence handler which can start a sequence customized in database. In every sequence I can read  or write tags, call database procedures and write the result of the procedure to the plc or as a byte stream to other ip adresses (eg: printers). The service has also a built in web server using Nancy, which can call database procedures, show websites, read the logging, read and write the customized plc tags. I use this service for data exchange between plc, database, label printers and for user display and interaction.
    
The actual plc connection uses OPC UA, which works fine, but it is a litte bit slowly with the on board       
UA Servers of the S7 1500. The S7 1500 (here a CPU 1507S) has also a disadvantage in the actual release: If you connect with TIA portal and change the code on the fly, the OPC UA Server stops and restarts after several seconds. So the service looses connection for aproximate 7 seconds. But some times the OPC UA server itself fails to start correct and I lose the connection to the projected tags of the plc. Here helps only a power off. The german AEG prinzip (Ausschalten , Einschalten, Geht) in english: power off, power on, works.

So I looked for a possibility to wide the range of the service and talk directly to the plc. Including the advantage of 
- talk to plc's without built in OPC UA Servers (like the  S7 300)
- get rid of the annoying OPC UA Server restart bug   
- availability of the source and the possibility to change it myself if there is a bug

S7NetPlus perfectly fits my needs.
 
In the customizing of the tag I need to fully declare thy type of the tag value. 
eg: bit, byte, int, word, dint, dword, real, string or stringex. And for a string/stringex the length.   
If I don't do this, a database failure, which send a to long string, could break the DB of the plc and overwrite values.  
    
Jürgen

mycroes, 2018-07-10:
Hi @bauerju 

I'm in a similar situation regarding data reading (by the way, can't you subscribe to changes with the builtin OPC UA? It's an optional protocol feature, I just don't know if Siemens implemented it). I have a `Register` entity with the following fields: Address (string), Type (string), Length (int). The type is converted to .NET `System.Type` (using `Type.GetType(register.Type)`.

I originally used this approach in combination with Siemens SAPI-S7 and only recently added support for S7NetPlus and my own library Sally7. We also read enum values (quite often they're the most important values even), which can be done that way as well (although Sally7 even supports them out of the box). I actually still use the addressing from SAPI-S7 because that allows us to use Sally7 in existing projects. Unfortunately without changes in my infrastructure that now means I parse addresses with a regular expression for every variable read or written, but even then it's blazing fast.

bauerju, 2018-07-10:
Hi @mycroes 
I use the "dataFEED OPC UA .NET Server & Client SDK for Windows" from Softing. I receive change events. But I can't say If they were created from the plc or from the client itself. I think it has an fallback mode...

I also have a conversion layer where I use a dictionary with connection information: which plc is used how to talk with it, tag address, tag type and value conversion.

Today I read my tag definition, extract the type and length and create a dataItem which is afterwards modified in Vartype and Count:
It is called eg 
- DB1.DBB1 -> 1 byte
- DB1.DBB1[20] -> 20 bytes
- DB1.DBB1[20].StringEX -> 22 bytes!, string length 20 chars
- DB1.DBD1:Real -> 4 Bytes as Real (Single)

In my opinion this should be better done in S7NetPlus than in my layer. Other could profit here. 
But this is your choice. I can only give you my ideas.
  

        private void CreateDataItemFromString(string variable)
        {

            string replacedType = "";
            if (variable.Contains(":"))
            {
                var splitted = variable.Split(':');
                replacedType = splitted[1].ToLowerInvariant().Trim();
                variable = splitted[0];
            }

            int count = 1;
            if (variable.Contains("[") && variable.Contains("]"))
            {
                int.Parse(variable.Substring(variable.IndexOf("[") + 1, variable.IndexOf("]") - variable.IndexOf("[") - 1));
                variable = variable.Substring(0, variable.IndexOf("["));
            }

            dataItem = S7.Net.Types.DataItem.FromAddress(variable);

            if (count > 1)
            {
                dataItem.Count = count;
            }

            if (replacedType == "real" && dataItem.VarType == VarType.DWord)
            {
                dataItem.VarType = VarType.Real;
            }

            if (replacedType == "int" && dataItem.VarType == VarType.DWord)
            {
                dataItem.VarType = VarType.DInt;
            }

            if (replacedType == "int" && dataItem.VarType == VarType.Word)
            {
                dataItem.VarType = VarType.Int;
            }
            
            if (replacedType == "string" && dataItem.VarType == VarType.DWord)
            {
                dataItem.VarType = VarType.StringEx;
            }
}

Btw.: I don't read classes or structures. This as very good concept for fixed programs.  You declare an structure like an UDT of the plc ad you can read and write at once. 

Jürgen

mycroes, 2018-07-10:
@bauerju the more I think about it, the more I feel this is not something that should be addressed by S7NetPlus. You're trying to use a string as object (i.e. combining multiple fields into a single container), which is usually not a best practice in programming. Also often it's way easier to maintain code if the logic is explicit, i.e. just store the VarType in a separate field so that when VarType is expanded you can use the new values without having to change your code.

Even with your model that allows dynamic forwarding of the data to different types of handlers, I assume you have constraints on the data the handler accepts. Let's take your printer example:
If you would read a real and send it to the printer, what will happen? I expect it'll throw an exception because the data type doesn't match. If that's the case, then you could've detected that error at an earlier stage by properly typing your data and you would've never even have attempted to read that data to send it to a printer.

I agree with your statement that it could be useful for other people as well, however I don't believe it should be part of S7NetPlus. You can of course create S7NetPlus.StringAddressing (or whatever pleases you) and offer it as a companion NuGet package. I'd be happy to help you out with setting up the repository for that, if you need any help with that.

 We're also in the process of deciding what the future for S7NetPlus will look like. Due to the complexity in string addressing (the issues I pointed out with different types, structs, enums) I envision it will be removed entirely in the next major release. That wouldn't change the fact that anyone could just wrap it around S7NetPlus anyway, but I'm not sure if that changes your view on the usability of this feature.

bauerju, 2018-07-10:
Hi @mycroes 
to your printer example. The tcp stream accepts only string or byte array as input, so the real would be converted to a string. But this doesn't make sense. This would not be enough information for printing. Normally I call a stored procedure in database which creates the byte array for printing. Here I can alos change the encoding of the string to byte array conversion. (some times the labelprinter need codepage 850) 

Yes, String support in plc is even not easy. I can hear it every time when I talk to the plc people. They don't like it ;-) But today with all the bar code readers you need it. 

I agree with you that string handling is complex. And If you plan to remove string support in S7NetPlus. Ok your decision. I must accept that and will then do my own wrapper that converts string to vartype byte. with correct length. 

Jürgen

mycroes, 2018-07-10:
Well string handling isn't complex, we'll never remove support for that. String addressing of variables is complex due to issues mentioned in this topic.

bauerju, 2018-07-10:
Hi @mycroes 
Then I have to apologize me. I misunderstood you. You think about removing the support for parsing the address out of the variables (DB1.DBB12). 
As long as Read(DataItem[])  and Write(DataItem[]) or something equal is supported ;-) I'll be happy.  
Parsing the name to create a DataItem is not a big problem. I could do this externally. 

Jürgen

mycroes, 2018-07-10:
@bauerju So can we close this issue as far as you're concerned? My offer to help you set up a repository will still stand of course (and you could also take a look at [mycroes/ReflEx](https://github.com/mycroes/ReflEx) for a very basic repository to get started quickly).

bauerju, 2018-07-11:
Hi @mycroes 
Yes I'll close it. Thanks.

PR #149 (2018-07-11): Add support for creating DataItem from string
Fixes #147.
closed 2018-07-11 by bauerju