S7NetPlus/s7netplus #148: Testers wanted! (S7-200 and S7-400)

mycroes, 2018-07-03:
To test the improvements and ensure compatibility we're looking for people with access (or who can provide access) to an S7-200 and an S7-400 PLC. We have access to S7-300, S7-1200 and S7-1500, but testers for those types are welcome as well.

If you have access to an S7-200 or S7-400 and want to help out, please let us know! Programming skills are not necessarily required.

JekRock, 2018-08-11:
Hi!
I can help you with testing the library on S7-400

JekRock, 2018-08-11:
@mycroes Okay, just text me if I can help!
By the way, I had problems working with S7-400 via this library. The connection is ok, but when I was trying to read or write something I had an exception without an error code. The same is for a similar library in different languages.

mycroes, 2018-08-14:
@JekRock Did you follow the instructions in https://github.com/S7NetPlus/s7netplus/blob/master/Documentation/Documentation.pdf with regards to setup in TIA portal? I'd guess it should work with S7NetPlus as well as with other libraries when the PLC setup is OK.

JekRock, 2018-08-15:
@mycroes we don't use TIA portal. Now we're getting access to plc via snap7 v. 1.3.0 and it works fine with the same plc setup.

JekRock, 2018-08-20:
@mycroes I've tried to read/write data again and everything works fine. Seems like something was not ok with plc.

mycroes, 2018-09-07:
@aetasoul wrote in [#185](https://github.com/S7NetPlus/s7netplus/issues/185#issuecomment-419365918)
> If you need any help to test the library, or improve it tell me. I'm interested in it and I have this cpu: 315-2 PN/DP, 315-2 DP, S7-1500, S7-1200, ET200SP, s7-314, 316-2 DP, 317-2 DP

breeze393, 2019-04-02:
Hello, I have problems with writing bits to s7-400 plc.
It returns "write failed with error 33153".
Reading the same bits - ok. 
Tried to write int - ok.
Also tried with simulation on PLCSIM, Winlc - everything is ok.
The pc where I tried to connect to plc also has Simatic.NET software with active connections.

Any ideas?

Hefaistos68, 2020-05-05:
If you still need, I have access to multiple S7-1200.

FalcoGoodbody, 2020-09-09:
I have access to S7-1500 and S7-300.

scamille, 2020-09-09:
@mycroes it might be possible to get all the Snap7 unit test working on a arbitrary Up/plc-type. That way this nice set of tests could be run against a exhaustive set of true Place.

mycroes, 2020-09-09:
The issue with the unit tests is that they rely on specific addresses. OTOH the main difference between the PLC types is in the connection setup phase, so just begin able to establish a connection already covers the most important part.

Either way, I guess @FalcoGoodbody already tested 0.7.0 against S7-300 and S7-1500? @aetasoul could you test S7NetPlus 0.7.0 against the ET200SP and S7-1200? Anyone else that has time and hardware to test, results are very welcome!

aetasoul, 2020-09-09:
@mycroes I leave tomorrow morning at 4 for the holidays and I return on the 22nd, as soon as I get home I run the tests

mycroes, 2020-09-09:
Thank you @aetasoul, enjoy your holidays!

FalcoGoodbody, 2020-09-10:
I have already tested with S7-1516, S7-1512SP (ET200SP CPU) and S7-300. That are the PLC types I have access here

mycroes, 2020-09-10:
Thanks @FalcoGoodbody!

aetasoul, 2020-10-03:
All ok on S7-1200