S7NetPlus/s7netplus #151: Release schedule proposal

mycroes, 2018-07-03:
A lot of changes have made it into S7NetPlus recently, so I propose we release sooner rather than later.

Proposed for **0.2.0**:
- [ ] ReadBytesWithSingleRequest cannot read >491 Bytes on S7-1500 #145
- [ ] v0.1.9 PLC ,The remote host forced the closure of an existing connection. #128

Targets for later release (not necessarily all at the same time):
- [ ] Calculate PDU size constraints #133
- [ ] Inconsistent specification of bit index #150
- [ ] ReadClass and ReadStruct order is not guaranteed #135
- [ ] Exception vs LastErrorCode / LastErrorString #137

rapha-dev, 2018-07-04:
@mycroes 
We should add PR #152 in next release, have added to proposal.

mycroes, 2018-07-07:
Moved this to the [wiki](https://github.com/killnine/s7netplus/wiki/Planned-releases). I'm not sure what's the best way to manage this, but I'm not a fan of keeping this issue open for months/years.

closed 2018-07-07 by mycroes

rapha-dev, 2018-07-07:
Maybe a project (tab) can be used to organize.

mycroes, 2018-07-07:
That sounds like a good idea.

rapha-dev, 2018-07-07:
Will create a new project without automation, can be added later when required.

killnine, 2018-07-07:
I have created the S7Net organization but haven't moved this project under it. Are you guys onboard with me moving it this weekend (today, perhaps)?

rapha-dev, 2018-07-07:
Sounds good, will be back in few hours and can do some tasks.

mycroes, 2018-07-07:
Yes, that would be great.