S7NetPlus/s7netplus #153: Allow override of source / destination TSAP

mycroes, 2018-07-05:
During COTP ConnectionRequest a source and detination TSAP are specified. These are calculated from the `CpuType`, but only a single connection can be established (at least from the same host, haven't tested multiple hosts) using the calculated TSAP's (and actually at least S7-1500 doesn't care about the source TSAP at all in this case).

It's possible to define additional S7 connections in TIA portal. To do so you must add a PC station with network connection and application component. The source and destination TSAP's that are generated by TIA portal can be used to establish additional connections (tested with Sally7, which does support this). The only downside of these configured connections is that they're also bound to the IP address that's specified on the network component of the PC station.

rapha-dev, 2018-07-07:
Another approach might be to implement a connection handling, like Siemens Prodave does (it supports 64   concurrent connections to driver). So the library must ensure consistent sends/receives and passing it back to originating connection, could be named _virtual connections_.
Unsure if this is achievable by using async communication to PLC.

mycroes, 2018-07-07:
I'm not exactly sure what you mean but I have a feeling it's about ensuring a single request goes out from S7NetPlus at a time. If that's the case, that's pretty easy. I have an `ExclusiveTaskQueue` that runs one call at a time and returns a task to await execution. I have actually wrapped both S7NetPlus and Sally7 in that way.

As far as performance is concerned, I made a test program doing the following (with Sally7):
- Read bool, byte, int, int , short, float, short, bool[10], string[2 + 20], byte[16] in one read request
- Write bool, byte, int, int, short, float, short, string[2 + 20]

This was then performed in a loop for 10 seconds and I tested with multiple threads / connections running the loop.

The results:

| num threads/conns | total cycles | cycles / s |
|-----------------|------------|---------|
| 1 | 2154 | 215 |
| 2 | 2334 | 233 |
| 3 | 2239 | 223 |

I actually ran this test up to 10 threads / connections, but those results were all pretty similar to 3 threads . connections. Also, I wrapped Sally7 in another class to make it compatible with our infrastructure, so actions include parsing a string address (using regex), creating a `DataItem<TValue>` (that's Sally7 specific) for every value, so probably that's the reason why 1 thread / connection is 'significantly' slower. Otherwise it seems bound by the PLC only (S7-1500, connected over 100MBit USB to ethernet adapter -> 5m cable, switch, 1m cable -> PLC).

So getting back to the issue at hand, for performance it's not necessary to use multiple connections. However, if you want to run multiple applications that connect to the same PLC you need to have separate connection TSAP's. I don't **need** this in S7NetPlus, since I use Sally7, but I think it's a welcome addition to S7NetPlus.

rapha-dev, 2018-07-07:
Yes, you got it right. Maybe it should stay at the user's responsibility to ensure single usage of s7netplus as it can be done by other libraries (i.e. your ExclusiveTaskQueue).
But the user cannot _forget_ to ensure this if it is already built in. Your performance comparison shows (which is very interesting btw.) no huge decreasing throughput on higher thread count, so it is not a _no go_, indeed.

mycroes, 2018-07-07:
I will consider adding a concurrent version of `Plc`. I have this planned for Sally7, but in Sally7 I have only a few methods that I need to take care of to enable concurrent use. In S7NetPlus it'll probably be a lot more work because there's a lot of code in different methods that actually accesses the `TcpClient`. You might have already spotted this in Sally7; all of the request buildup and response parsing is in separate methods, so adding a sync version or a concurrent version doesn't take much effort.

mycroes, 2018-07-07:
@rapha-dev For your pleasure: [`ExclusiveTaskQueue`](https://gist.github.com/mycroes/c4f23b056494fa844ad071f1c10782ca).

mycroes, 2021-06-04:
Taking care of this in #396.

closed 2024-04-28 by mycroes