S7NetPlus/s7netplus #157: Snap7 server only supports reading 20 dataItems in a single request

mycroes, 2018-07-07:
Snap7 server only supports read of up to 20 dataItems in a single request. I'm not sure why they chose this, but I did email the author of Snap7 to see if I can get an explanation.

Although at first this doesn't really seem to matter much, it doesn't allow us to create unit tests for more than 20 `DataItem`s in a single request. One way to fix this might be to add server support to S7NetPlus or Sally7, but honestly I think Sally7 is the easier candidate here since all the structs are already mapped.

rapha-dev, 2018-07-09:
So Snap7 limits this in server support? Then we should add server support or, as you proposed, Sally7 (server) as we can get more than 20 DataItems out of PLCs (except 240b PDUs).

killnine, 2018-07-10:
I think @mycroes is saying it'd be easier to add server support to Sally7 right now than to S7N but it doesn't exist yet, correct?

What's the use case for this? Is it causing testing challenges?

rapha-dev, 2018-07-10:
@killnine It will affect UnitTests. Currently max. 20 DataItems can be passed to Snap7 server, S7 protocol (and S7-400, 1200 and 1500) can handle more. S7netplus should allow maximum possible DataItems and so a UnitTest should proove that (PDU constraints).

killnine, 2018-07-10:
I'm just throwing it out there that it'd be nice to have no external dependencies.... 😄 

But maybe that's something for a future release.

thoj, 2018-08-16:
This is the same issue I encountered. Tests with real cpus with more than 20 data items work fine if the CPU negotiates a higher pdu size. I suggest that we just use the negotiated pdu size and only enforce the 20 dataitem limit on snap7 tests.