S7NetPlus/s7netplus #159: Incorrect String handling with vartyp StringEX

bauerju, 2018-07-09:
Hi. Writing a string over a daitem, declared as StringEX,  doesn't work. 

The first two bytes in then memory, which represents declared and used length are overwriten.  
Simple Code:
[image]

DB before writing:
[image]

DB after writing:
[image]

Jürgen

mycroes, 2018-07-09:
I believe the current code doesn't check the vartype when writing. I'll try to fix it this evening.

bauerju, 2018-07-09:
Hi @mycroes 
Yes you'r right. I digged a bit into the code. There is no check for  the vartype. 

Jürgen

mycroes, 2018-07-09:
@bauerju / Jürgen thanks very much for your feedback, address this issue in PR #162. I'm very happy with the way you provide feedback, so please keep it coming.

PR #162 (2018-07-11): Fix write stringex
Fixes #159.
closed 2018-07-11 by rapha-dev