S7NetPlus/s7netplus #164: Incorrect String handling with VarType.String

bauerju, 2018-07-11:
Writing VarType.String doesn't care about the length of the string.:
[image]

Plc.Write(da) writes full string length into the plc. 
[image]

Plc.Read() cares about length and reads only correct length.
[image]

Jürgen

bauerju, 2018-07-11:
HI
Writing a shorter string, which is shorter as defined, writes only string length.
eg. 
string in DB is defined as 8 bytes and value is `ABCDEFGH`
Reading gives correct  `ABCDEFGH`
Writing `0123` , 
string in DB is after write `0123EFGH`.
Should the rest not filled with \0 ? I Think it should be `0123\0\0\0\0`

Jürgen

mycroes, 2018-07-11:
HI @bauerju. Agreed in full. I will change the implementation to write at most `count` bytes and to pad with `\0`.

PR #166 (2018-07-11): Fix string to byte array
Fixes #164
closed 2018-07-11 by mycroes