S7NetPlus/s7netplus #165: Reading bits - length parameter not the same

arnoraps, 2018-07-11:
If I read bits with this statement:
`var bits = (BitArray) plc.Read(DataType.DataBlock, 1, 1, VarType.Bit, 2, 0);`

Than I get 16 bits in a BitArray.

If I read bits with this:
`var bits = (BitArray) plc.Read(DataType.DataBlock, 1, 1, VarType.Bit, 1, 0);`

I just get a single bit (true/false) With or without casting to Bitarray makes no difference.

rapha-dev, 2018-07-11:
@arnoraps 
Thanks for reporting.
The first `Read` should read 2 objects (`DB1.DBX1.0` and `DB1.DBX1.1`), will have a look at.

killnine, 2018-07-11:
In `ParseBytes`, it checks what the number of elements you want to return is...

```
 private object ParseBytes(VarType varType, byte[] bytes, int varCount, byte bitAdr = 0)
        {
            if (bytes == null || bytes.Length == 0)
                return null;

            switch (varType)
            {
                case VarType.Byte:
                    if (varCount == 1)
                        return bytes[0];
                    else
                        return bytes;
                ...
            }
        }
 ```

What this means is that the return type is a byte array in your first example and a single byte in the second. Not great behavior, though.

PR #167 (2019-07-17): Fix incorrect length when reading BitArray
Fixes #165
closed 2019-07-17 by mycroes